bozimmerman/Zimodem #2: Soft on ESP8266 MCU V2 - freezing

thErZAgH, 2017-05-24:
Hi Bo,

Thanx for the very nice software for ESP8266/C64/communitiy. I installed the soft on ESP8266 MCU V2. With Arduino. It works. But if I do BBS or downloads from Commodoreserver.com it's freezing... I know you put the soft on Adafruit module. But: is it able to run flawlessly on native ESP8266 MSU V2? Other wifi soft is running /wout problems on this module.

Any suggestions?

Best regards.

thE rZA

thErZAgH, 2017-05-25:
Hi Bo.

It's still freezing with 2.61.

bozimmerman, 2017-05-25:
Sorry, I sent you a response in email, but I guess it didn't go through.

I would need some more information to find out why it is 'freezing'.

For example, I don't know what 'Adafruit' is.  I connect to telnet BBSes all the time, so you must be doing something specific.  What program are you using?  What commands do you enter?  I also don't know how you do downloads from Commodoreserver.com. 

If you could give me any specific steps to cause the freeze, I'm sure I can help.

- Bo

thErZAgH, 2017-05-25:
Hi Bo. Thanks for comming back. Didn't read Mail. Maybe it's in spam folder... Have to look at it later...

No. I'm not doing something special. What I did:

* Load "V-1541.2400" driver from commodoreserver.com (from d64 or as prg)
* Then SYS49152 (start driver)
* Type new then SYS49152 again
* Load a simple basic program for ATD"commodoreserver.com:1541" (on 2,2,2)
* If connected to Cserver quit basic program
* LOAD"$",2 (works)
* LOAD"*",2 freezes after two block loading...

Other ESP8266 driver like 1200baud (alwyz code and devirates) are working fine there... I realy like to try your driver because of IRC and FTP clients.

And if you are there: can you try V-1541.39k driver, too? And: BBS at 9600 misses some chars, too! Below 9600 (1200, 2400  working fine!).

Thank you!

bozimmerman, 2017-05-25:
Hello! Wow, so, the modem designed for Commodoreserver does not use the Zimodem firmware.  So, the reason it didn't work is because it wasn't even designed to work.  The fact that LOAD"$",2 works at all is simply AMAZING!

In fact, I find your experiment so exciting that I will make it my goal to find a way to make it work.

I'm guessing the reason 9600 misses characters is because Zimodem (and our hardware) does not support hardware flow control.  however, we are adding that ability to our modem soon, which means Zimodem will also support it soon.

So, I'm going to modify your ticket into a Feature Request, and work on it as soon as I can.  Very interesting!

thErZAgH, 2017-05-26:
Hi Bo.

Thank you for reply. Yeah. It would be great to see that your soft can do Commodoreserver.com stuff. I'll test it for you. I know WiModem can do Cserver up to 38.4kbd. But this is closed code :(. Driver of Cserver can 2400bd and 38.4kbd. No 9600 driver there :(. 

The 1200baud "Alwyz" driver can do Cserver with 2400bd, but not with 9600bd :\. You can find ESP driver here https://1200baud.wordpress.com/2017/03/04/build-your-own-9600-baud-c64-wifi-modem-for-20/ or/and here https://github.com/jsalin/esp8266_modem .

What realy would be great: that your open source(!!!) code can do 2400bd, 9600bd AND 38.4kbd on Cserver!!! That would freaking great.

I would test it out for you and spread the word on german c64 forum - if it works. Many ppl here are interested in doing BBS (and they often don't now what's possible except BBS)...

Marco
thE rZA

bozimmerman, 2017-05-27:
Hi, so I tried out v-1541 (2400 baud) and cometchat.  It's amazing how well it works.  I fixed a buffering issue in the Zimodem firmware, so you may want to enter AT&U6502 to get the latest firmware.

Anyway, I was able to load programs, load directories, be in the cometchat area, everything! It's pretty awesome.  The next thing I will do is try out other baud rates, if I can figure out where those drivers are.  This is pretty neat!

I will add a little BASIC program to the software package that will configure the modem for 2400 baud and connect it to commodoreserver.com, and maybe load the v-1541 driver also and activate it.

thErZAgH, 2017-05-27:
Hi Bo. 

A quick test was successfull. But what you shouldn't do: press modem reset after AT&U6502 emmidiately. Like the Info shows in terminal. After that you have a corrupted firmware. Had to reflash modem via Arduino again :). Maybe it was too quick after message ....

thErZAgH, 2017-05-30:
I just tried V-1541.2400 another time. Sometime it's freezing (again)...

bozimmerman, 2017-05-30:
Hey.. whenever it freezes, try to write down exactly what commands you entered, and in what order, so that maybe I can figure out why.

bozimmerman, 2017-07-08:
So, I have not even been able to connect to commodoreserver.com for many weeks now -- at ANY baud rate.  Does it work for you?

thErZAgH, 2017-07-08:
Hey Bo.

Let me give it a shot tonight.

bozimmerman, 2017-07-10:
Ah, it looks like it's back up today.  Perhaps that email I fired off to him was helpful. :)

closed 2017-09-09 by bozimmerman