bozimmerman/Zimodem #3: Uhm.How does it work?

thErZAgH, 2017-05-26:
Hi Bo.

Sorry to nag you again. But: I set the baudrate of Zimodem to 1200. Then I loaded pml64.bin. After that I started one of the PRG files. IRC for e. g. But nothing happens. You wrote to reset cpu and do all again. It changes nothing.

Must I disconnect 9600 cables to make it work? BTW: in CCGMS I have to change baudrate only and it works....

bozimmerman, 2017-05-27:
I don't know what the 9600 cables are, but if you mean disable UP9600 compatibility, no -- you don't have to do that.  However, UP9600 connections are not compatible with the C128, either in C128 mode OR in C64 mode.  As you can tell by this response, I would need a lot more details before even guessing why it's not working.  Do ANY of the Zimodem/C64Net tools work? Configure? telnet? ftp?

thErZAgH, 2017-05-27:
Hi Bo.

Thanks for comming back, Yes. Any of the tools are not working. Here's what I'm doing:

* start CCGMS
* set 1200 bd with ATB1200
* write 1200 bd with AT&W
* result: ok
* warm reset C64
* LOAD"PML64.BIN",8,1
* LOAD"TELNET64-128",8,1
* screen with info "TELNET v1.0 Requieres .... Initializing modem....." comes up
* ... nothing - neither after two seconds nore after several minutes.

This happes with ALL of the basic programs. And yes: I did't cut the up9600 patch cords. I'm using C64C with 1541 II. Not a C128 or similar...

bozimmerman, 2017-05-27:
OK, first off, you should do LOAD"TELNET64-128",8 and enter RUN.  Do not load it ,8,1.  And do not load"PML64.BIN" -- the telnet program will do that for you automatically.

The telnet program is one of the simplest programs, since it does not switch the modem to 9600 baud, or use packet-mode, so I'd like you to keep trying the telnet program.

You are using 2.61 (or maybe now 2.62) right?

If it still doesn't work after the changes I mention above, I would like to start running some BASIC experiments to figure out why.  But first, please change how you load and run the program.

bozimmerman, 2017-05-27:
OH! I wonder if V-1541 is conflicting with PML64 -- they both reside at $c000.  I will make sure I change the telnet program to better detect when V-1541 is installed and disable it before loading PML....

thErZAgH, 2017-05-27:
I think there are v-1541 lo drivers, too. They have other entry points, I guess.

thErZAgH, 2017-05-27:
Hi Bo. 
If I first load Telnet64-128 without doing anything else before and type RUN it will load something from tape!? "Press Play on Tage"... On ,8,1 or ,8

bozimmerman, 2017-05-27:
LOAD"TELNET64-128",8

bozimmerman, 2017-05-27:
RUN

thErZAgH, 2017-05-27:
"press play on tape"

bozimmerman, 2017-05-28:
I see.. it DOES sound like v-1541 was interfering, and that the modem was still in 2400 baud mode.  In other words, there were several problems all happening at once, and any one of the problems would be enough to break it.  I definitely need to detect v-1541.

bozimmerman, 2017-05-28:
Also, if you type in LOAD"TELNET64-128",8 and your computer responds with "PRESS PLAY ON TAPE", then your computer has some sort of problem. :(

thErZAgH, 2017-05-28:
Hi Bo.

I haven't loaded the V-1541 driver. I'm using the D64 disk provided in the repo. 

If I patch TELNET with OPEN5,2,0,CHR$(10) and LOAD"PML64.BIN",8,1 it steps over the initializing message. If I want to call BBS then (coffemud) I get this exception:

http://imgur.com/a/U9wtL

BTW; can we have a chat in IRC or somewhere?

bozimmerman, 2017-05-29:
I fixed that exception .. please try 2.63

thErZAgH, 2017-05-30:
Hi Bo.

I know that $(10) is for 2400 baud. BUT MY MODEM IS SET TO 1200 baud!!! 

Call me wrong, but after loading PML64.BIN software ist getting data with 2400 baud speed?

bozimmerman, 2017-05-30:
Yea, sorry about 2.62 -- I put a pretty nasty bug in it that I didn't find until later.

bozimmerman, 2017-05-30:
Are you in Europe, BTW? Using a PAL C64?

thErZAgH, 2017-05-31:
Hi Bo.

Yes. I'm in Europe and using a PAL C64.

bozimmerman, 2017-06-01:
Do you think you can help me make the software work better for PAL users?
For example, in the TELNET64-128 program, on line 30, you see this: poke 665,73:poke666,3
Those pokes are "corrections" to the Commodore 64 KERNAL to allow it to get 1200 baud without errors.  Do they work OK for you? Do you see bad characters when using that program?

thErZAgH, 2017-06-04:
Hey Bo, Sure. I can do. I'll test it t'night...

thErZAgH, 2017-06-18:
Hi Bo.

Nope. Doesn't work for me. I have to change

15 OPEN5,2,0,CHR$(8):DIMPP$(25):P$="ok":POKE186,PEEK(254)
to 
15 OPEN5,2,0,CHR$(10):DIMPP$(25):P$="ok":POKE186,PEEK(254)
(without this initializing is freezing)

and

40 IFSY=226ANDPEEK(ML+1)<>209THENCLOSE5:LOAD"pml64.bin",PEEK(186),1:RUN
to
40 IFSY=226ANDPEEK(ML+1)<>209THENCLOSE5:LOAD"pml64.bin",**8**,1:RUN
(without this C64 will load from TAPE)

bozimmerman, 2017-06-21:
Well, I set up my PAL C64C and tried out one of the programs.  You are right, you have to change it to chr$(10).  That is the craziest thing I've ever seen.  Even more so because almost immediately the program, written for 1200 baud, tweeks some of the timing registers to better match 1200 baud.  I clearly have much research to do.

bozimmerman, 2017-06-22:
I have the official fixes for you!  I will try to upload a fixed disk with all of these changes today:
1. Keep the chr$(8)... it really is necessary.
2. On line 30 of almost every program you see POKE665,72.  Change that to POKE665,43.  That is the proper PAL value.  The version I release will check PAL/NTSC first, of course.
Everything else should work fine. Let me know if you still have problems.  That change fixed the programs for my PAL C64C running on 110V at 60hz.

bozimmerman, 2017-06-22:
Oh, change all the loads that use peek(186) to use peek(254) instead .. doing that also.

thErZAgH, 2017-06-22:
Hey Bo,

I changed POKE665,73 to POKE665,43 and peek(186) to peek(254). This seems to work WAY BETTER then before 👍 . IRC CHAT v1.0 and TELNET v1.0 are working now.

So I'm ready to test the whole new D64 if you have uploaded it to gh.

bozimmerman, 2017-06-23:
OK, I've updated the .D64 image and the apps in github -- they should all be PAL compatible now.  Let me know if you have any troubles.

closed 2017-06-23 by bozimmerman