bozimmerman/Zimodem #4: Request: IEC Display

ILAHWWINC, 2017-06-27:


ILAHWWINC, 2017-06-28:
I would like to have a little display to show some status information.

Maybe like the one in this firmware (https://github.com/SuperIlu).

Is it possible to include the code? Do you think it woll be helpful?

thErZAgH, 2017-06-29:
Hey ILAHWWINC.

What display do you think of?

ILAHWWINC, 2017-06-29:
Hey thE rZA,

i think about an OLED 128 x 64 I2C Display. Connected to GPIO 0 and 2. (SDA := GPIO0, SDC := GPIO2).
With the SuperIlu Firmware (forked from RolandJuno/esp8266_modem) there is inforamtion about WiFi - Connection, Transfer Speed, IP - Adress Connection and Duration.

thErZAgH, 2017-06-30:
...

bozimmerman, 2017-07-08:
Ah, but the modem firmware requires the use of those pins for DCD and RTS/CTS flow control.  Maybe it would work with one of those big nodemcu esps that have like 10 i/o pins.

ILAHWWINC, 2017-07-09:
A version with display for NodeMCU would be fine. The costs for this unit are very low. I use the hardware from 1200baud.wordpress - modem for 10$.

---
Esp8266 GND -> C64 User Port Pin N (Gnd)

 TxD (GPIO1/TXD0) -> C64 User Port B and C (RxD. Yes, Short B and C together)

Esp8266 RxD (GPIO3/RXD0) -> C64 User Port M (TxD)

Esp8266 GPIO4 -> C64 User Port K (CTS)

Esp8266 GPIO5 -> C64 User Port D (RTS)

Then:

Add a wire from M to 5 on user port plug
Add a wire from B,C to 7 on user port plug
Add a wire from L to 6 on the user port plug

Optional power from C64: Connect C64 pin 2 (+5v) to Esp8266 
---
Maybe there is a chance to get this hardware and your firmware together?

bozimmerman, 2017-07-10:
BTW, the latest version of the firmware supports using GPIO4 for CTS (RTS should have just been held High).  You can take advantage of all the firmware features if you connect DCD by running a wire from user port pin H to GPIO2.

genesis8bit, 2019-03-01:
Would Zimodem work with an Odroid Go (ESP32 inside) which includes a LCD screen with a 10 pin expansion port (I2C, GPIO, IRQ at 3.3Volt) ? More details on the pins at https://wiki.odroid.com/odroid_go/odroid_go

bozimmerman, 2019-03-04:
As I don't have this hardware, I couldn't say.

kikendo, 2019-07-18:
Came looking for this... maybe a simpler display like a 2 digit LED could also be added or would be simpler to add? It could display short messages for status

bozimmerman, 2019-11-16:
If someone is willing to send me hardware, I'd be happy to take a look at doing this.  If that happens, please re-open the issue.

closed 2019-11-16 by bozimmerman