bozimmerman/Zimodem #7: Unable to build version 3.1

genetoye, 2017-08-19:
I downloaded the source for version 2.0 back in March and successfully built it with Arduino 1.8.1 with the 2.3 version of the esp8266 board code (appears to be current version from their GitHub).  However, attempting to build the 3.1 version results in compile errors.  The first was in wificlientnode.ino when the following code was used:
int WiFiClientNode::flushOverflowBuffer()
{
  if(overflowBufLen > 0)
  {
    int bufWrite=client.write(overflowBuf,overflowBufLen);

Use of the unsigned char array in client.write causes the following errors:
Arduino: 1.8.3 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DOUT, 115200, 1M (64K SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
In file included from C:\Users\Gene Toye\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39:0,

                 from sketch\pet2asc.h:16,

                 from C:\Users\Gene Toye\Documents\Arduino\Zimodem-master\zimodem\zimodem.ino:41:

C:\Users\Gene Toye\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h: In instantiation of 'size_t WiFiClient::write(T&, size_t) [with T = unsigned char [256]; size_t = unsigned int]':

C:\Users\Gene Toye\Documents\Arduino\Zimodem-master\zimodem\wificlientnode.ino:195:57:   required from here

C:\Users\Gene Toye\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h:123:36: error: request for member 'available' in 'source', which is of non-class type 'unsigned char [256]'

     size_t left = source.available();

                                    ^

C:\Users\Gene Toye\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h:127:5: error: request for member 'read' in 'source', which is of non-class type 'unsigned char [256]'

     source.read(buffer.get(), will_send);

     ^

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Commenting out the code except the return in flushOverflowBuffer function will resolve this error but new ones occur.

What version of the board support is required for successful compilation of the 3.1 version?

Thanks for a wonderful project.
Gene

bozimmerman, 2017-08-20:
Hey genotoye --

Well, the good news is that, worst comes to worst, removing that code is fine: I've never been able to trigger it's usage.  The point of it all was somehow detect that the underlying socket code was not permitting any more bytes to be written, and that therefore the host computer needs its CTS line driven low to stop incoming data.  However, as I mentioned, I've never been able to get the underlying socket code to reject write data, no matter how lax the receiver is, how small the receivers buffers are, etc.

That said, I suspect your problem is that you are official the official library distribution (2.3.0) instead of the development head.  I had to switch to using the development branch very early on due to ridiculous and unusable write times, which were fixed in dev.  The README should give you some tips on how to replace your libraries with the latest development branch.  And that should fix your problems.

- Bo

genetoye, 2017-08-22:
Thanks for the answer.  That was exactly the issue.  I read the note in the readme but incorrectly assumed since the 2.0 built ok, I didn't need to update.  Following the directions allows the code to correctly compile.

closed 2017-08-22 by genetoye