bozimmerman/Zimodem #8: nnection.

dedhed, 2017-09-07:
Hello!  Very cool project, thank you for this.

I am using a c64net device and the auto updater brought me to firmware 3.2 no problem.  I am trying to telnet (atdt) to a Linux server and it looks like there is a problem with Telnet protocol negotiation.  I haven't looked at the source code for Zimodem but I was wondering if the Telnet implementation is RFC based or is it more like netcat streaming without a (or with a minimal) Telnet application layer?

I have attached a network trace (Wireshark) that shows where it hangs right after connecting...  the server seems to be trying to negotiate some Telnet settings.

Any help is appreciated.  Thanks so much!

dedhed, 2017-09-07:
[telnetcap.zip](https://github.com/bozimmerman/Zimodem/files/1283406/telnetcap.zip)

bozimmerman, 2017-09-07:
Hi.  Sorry to hear you are having troubles.  First, some information on what the telnet client code does:
1. It keeps garbage characters generated by telnet negotiations from being forwarded to the serial port.
That's it.  Oh, it also responds to the TERMTYPE code.  But that's it.
So, when the socket is in telnet mode and reads an IAC, it will immediately do a blocking read for the code.  Any WILL X or WONT X command it gets, it always responds with DONT X.  Any DO or DONT, it just ignores. A Subcode is parsed to see if it is TERMTYPE, and if it is, it responds, but otherwise ignores the subcode.
I'll now go take a look at your capture and see if something stands out.

bozimmerman, 2017-09-07:
So, I looked at the capture, and I see the DO This and DO Thats in the data.  There's no response, but given the algorithm I described above, I wouldn't expect any.  Does the RFC say that a client must respond to all DO commands, with, WILL or WONT I'm guessing?

dedhed, 2017-09-07:
If I am reading this correctly, it does look like the RFC expects at least a confirmation?

From RFC854: 

      DO (option code)    253    Indicates the request that the
                                 other party perform, or
                                 confirmation that you are expecting
                                 the other party to perform, the
                                 indicated option.

bozimmerman, 2017-09-07:
I'll take a look at the rfc and see if the client MAY respond to that command, or SHALL respond to that command. :)  May I ask what sort of server you are trying to connect to?  That might tell me a lot.

dedhed, 2017-09-07:
Thanks!  If I can help is any way at all, please let me know.  

I have tried two servers, the trace comes from a Raspberry Pi running Raspbian 8.  The other server is a VM running Ubuntu Server 16.04.2 LTS.

Thanks for the quick response tonight!

bozimmerman, 2017-09-07:
Heck, hang around another 30 mins, and I'll probably have a 3.3 for you to try out.

dedhed, 2017-09-07:
Haha!  I'll be here :-)

bozimmerman, 2017-09-07:
OK, I put 3.3 out there, but won't make it official until it's passed both of our tests.
To force your modem to upgrade to a version that's not official yet (like 3.3), you need to enter this:
AT&U=3.3

dedhed, 2017-09-07:
Both platforms are running telnetd (xinetd/in.telnetd) ... the other major option would be inetutils-telnetd from what I'm seeing.  Just in case you wanted it more specifically.

dedhed, 2017-09-07:
Ah just saw your last reply, sorry.  I'll give it a shot and let you know.

dedhed, 2017-09-07:
Woooot!  That did it.  9600 baud and flow control seems great too.

I'm totally floored by your response to this.  THANK YOU, amazing work.

bozimmerman, 2017-09-07:
Hey, you helped me find a crippling bug.  You have my thanks as well.  I only tested telnet against, wait for it, another telnet server I did that was "roughly inspired" by the RFC.

closed 2017-09-07 by bozimmerman

dedhed, 2017-09-07:
I'll let you know if anything else comes up :-)

Thanks again.