bozimmerman/Zimodem #11: Zimodem, ESP not responding : wifi.disconnect() and more ?

Sassa0, 2017-10-20:
Hello Bo,
Your project seems great !

For the past few weeks, I've been trying to build the firmware part on an ESP02 on a breadboard.

Here are the versions that I tried :
 - build platform : linux
 - arduino : 1.8.5 or 1.6.7
 - ESP8266 core for Arduino : latest git from 3 weeks ago, and from today (that uses SDK v2.1.0-10-g509eae8)
 - Your zimodem code from 3 weeks ago and from today

I've followed your indications and it builds OK and uploads OK to the controller.
As indicated, I put the serial console to 1200 baud.

However :
 - using the unmodified code, all I get is the usual "non-readable boot output" from the ESP but nothing after that. And it does not respond to AT commands.
 - when adding the following line before all of the 3 occurrences of "WiFi.disconnect()" in the code, then I get the init message, and it responds to the ATI command, but it does not lists the access points and does not connect with ATW, and freeze.
```
if(WiFi.status() == WL_CONNECTED)
  WiFi.disconnect();
```

I am somewhat troubled that I have to do this modification ?! I read that people have the same freeze with WiFi.disconnect() with other projects...

About the wireless not working, to be sure that it is a power issue, it is powered using a small "lab" power supply with controlled volts and amps. And I have tested that the ESP is OK with NodeMCU/Lua. I fear it is a software problem like the previous one ?

Can you please provide the software versions that you currently use so that I try them ?

Thank you.

Sassa

bozimmerman, 2017-10-23:
ATM, I've lately been building using the rev dated 9/2/2016, but just to be certain, I downloaded the latest as of the time of this reply, build the firmware, uploaded it, and tested WiFi scanning and connection.

Are you using Arduino 1.6.11? Could it be the way it is calling the compiler, or some incompatibility with the other Arduino libraries?  I'm afraid I'm just not seeing a problem.

I've tested this firmware with the ESP-01 and the NodeMCU hardware without problems.

Sassa0, 2017-10-28:
Hello,
Thank you for your help.
I reinstalled a windows pc with all the latest tools, but with no more luck on the same esp03 (not esp02 as stated in my first message). So I picked an esp01 from one of my past projects, and it works with your code, unmodified.
I think that my esp03 is broken... I'll try with an esp12 to be sure and update/close the issue when I have the time to try it.

So i'm very sorry that I took some of your time because of that...

bozimmerman, 2017-10-29:
No problem.  I'm glad you've figured it out!

closed 2017-10-29 by bozimmerman