bozimmerman/Zimodem #12: Basic Programs unable to connect to modem on a C128 PAL

hbanko, 2017-12-16:
After starting configure on C128D PAL in C64 modem I am getting after 2 minutes:

```
Intialization Failed
Is your modem set to 1200b?
break in 220
```
All the other programs on the disk are not working either. They either hang (Telnet) or through the same error. The modem is set to 1200 baud and works well in CCGMS in C64 mode.

I tried in the C128 mode as well and have the same issue. Besides the Basic programs not working I noticed that the transmission above 1200 baud gets quite unreliable.

bozimmerman, 2017-12-24:
Hello! Thanks for contacting me.  So, let's start with the basics:
1. What version of the firmware are you running? (You can get that in CCGMS by entering ATI)
2. What version does the configure program report at the top?

It certainly does sound like there is a 1200 baud timing issue.  That is a known bug in the C64 kernal.  However, my dadgum programs are supposed to fix that on line 30 of the configure program.  The C128 kernal, however, does NOT have this issue, so it's confusing that you see it there also.  Very confusing.  In C128 mode, you are using your computers built in kernal timing.

Which gives me an idea for another thing or two to try.
Thing 1:  Reset your computer to C128 mode.  Load Configure.  Instead of entering RUN, enter:
 RUN 50000
Now start typing AT and pressing enter.  It should respond with OK.  Enter ATI and press enter.  You'll get the little intro message.  All this is to test whether basic 1200 baud support is working in the stock C128 kernal.

Thing 2: Reset your computer to c128 mode.  Load configure.  Now change the following line by re-entering it:
209 IFCT<10THENGOSUB900:PRINTP$:IFP$<>"ok"THENCT=CT+1:GOTO207
(the only change is adding printp$:    )
Now enter RUN.  You will probably get the same error, but at least you might also get some feedback on what the modem is responding with.

hbanko, 2017-12-26:
Hi Bo,

Thanks for getting back to me!

> 1. What version of the firmware are you running? (You can get that in CCGMS by entering ATI)

ATI in CCGMS tells me its version 3.3

> 2. What version does the configure program report at the top?

Configure tells me it is version 1.4

After that I reset my computer into C128 mode (power cycle actually) and did the other two checks.

Thing 1: No response from AT or ATI at all.

Thing 2:
Same error "Initializing Modem...Initialization failed. Is your modem set to 1200b? Break in 220" and nothing else. Checked it twice and 2nd time I also added a PRINT CT. That also did not give any insights.

hbanko, 2017-12-26:
One thing I want to add and not sure if it is relevant. In CCGMS I have to configure the modem to UP9600 / EZ232. It does not work when I chose any of the other types (e.g. Userport 300 - 2400) and I have built it by using these instructions: https://www.c64-wiki.de/wiki/Wifi-Adapter

bozimmerman, 2017-12-26:
Ah, well, that cleared up everything.  I thought you were using a C64Net WiFi modem...
The C128 user port lines used for UP9600 conflict with the certain serial IEC lines.  This creates problems.  Frankly, I'm surprised your computer will boot with UP9600 hardware in a C128.  Mine doesn't (IEC locks up), which is why our modem has a switch to disable UP9600 for C128s.

So, if you plan to use this on a C128, my recommendation is to clip the UP9600 lines.    Since I know a bit more about your hardware, I'd also recommend hooking up pin H (DCD) to pin D2 on the ESP, instead of pin K.  It will make the programs you are trying to run work more correctly, and also allow you to run BBS programs.

closed 2017-12-26 by bozimmerman