bozimmerman/Zimodem #13: AT&W not saving baud rate

drencorxeen, 2018-02-05:
I did a 
ATB115200
I then did a
AT&W
but ZiModem is not saving the baud rate value on the ESP8266.

bozimmerman, 2018-02-06:
I couldn't recreate this.  I want to mention that we've seen some ESP-01s that, for some reason I don't understand, don't work with SPIFFS.  I would recommend trying some brief little sketch that does nothing but format the SPIFFS, write a file to it, and read it back.  I suspect you might have one of the defective ones.

allenhuffman, 2018-02-06:
drencorxeen - are you seeing "INITIALIZED" before it prints READY and OK on startup? I have been running fine on the ESP devkit, but after building for the GENERIC ESP8266 module, I find I cannot write config information. Thus, no baud rate, etc. I also cannot connect to WiFi but it shows them to me, so that may be a different issue.

allenhuffman, 2018-02-06:
Ah! I built mine with "NO SPIFFS" (default in Arduino IDE) and that would explain why I have no file system.

See Tools->Flash Size

I need to find out what mine should be set to.

allenhuffman, 2018-02-06:
There is an ESP8266 example Sketch in the Arduino IDE called "CheckFlashConfig" that might help us. Mine shows:

Flash real id:   001440E0
Flash real size: 1048576

Flash ide  size: 524288
Flash ide speed: 40000000
Flash ide mode:  QIO
Flash Chip configuration wrong!

Not sure what that means tho.

allenhuffman, 2018-02-06:
I changed my settings until I got it to match. Above, it seems like i have 1MB. I changed it to "1MB (64K SPIFFS)" and now I work. Saves and everything.

drencorxeen, 2018-02-06:
@bozimmerman ,

Thank you for informing me that I needed the SPIFFS.  I didn't know I had to select that.
I had it set to 512K without SPIFFS and my ESP8266-01 has 1MB of flash.  So I set it to 1MB with 512K SPIFFS and it seems to work fine.

Thank you. :D

bozimmerman, 2018-02-06:
Thanks for letting me know I had a hole in my build instructions -- I will make sure that's mentioned.
I'm glad it got worked out, though be aware that is a batch of ESP-01s out there which, even with the proper build settings, is failing to write.  We could not differentiate between them and any others -- it was quite frustrating.
Anyway, happy computing...

closed 2018-02-06 by bozimmerman