bozimmerman/Zimodem #14: WIFI credentials charset problem?

SCUBA82, 2018-02-19:
Hi Bo, 
I've just assembled my modem (with your firmware) this weekend and had an issue to connect to my Wifi. 
Lets say my SSID is "**Abc2**". 

I've used your "CONFIGURE" tool which initilizes the modem and searches for available Wifi SSIDs. 
The "CONFIGURE" tool shows my SSID as "**aBC2**". (the upper lower case syntax of other text is also messed up)
After providing the password, the connection attempt fails. 

Afterwards I've switched to CCGMS and entered the configuration mode with "AT+CONFIG". Trying to connect via the command "WIFI" leaded to the same result. 

After switching to "PETSCII translation on" the modem connected without problems and also the SSID was listed correctly as "Abc2". 
Now I had to deactivate the "PETSCII translation" again cause your tools doesn't seem to initialize the modem with this option activated.

What I've tested so far: 2 different C64 (PAL) machines (with sd2iec), JIFFY, CMB, SPEEDDOS, password in different "upper-lower case" syntax.

thank you for your effort!

bozimmerman, 2018-02-19:
Hello!

The Configure tool should have done what you ended up having to do by hand, namely handle all the text in PETSCII translation.  However, I have to admit I've haven't tested it with 3.4, which I guess you are attempting to use, since you mentioned trying AT+CONFIG

So that means you get a warning! Github, and thus 3.4, is basically the dev head.  If I'm in the middle of something and get sleepy, I check it into the dev head.  If something is broken and I want to fix it later -- to the dev head it goes.  If I add some new feature and want to test it later... well, you get it.    That means using the firmware directly off github is Dangerous(TM) to happiness. </warning>

BUT, there is a solution, though you may not like it -- you can Revert to the official, tested, stable, full-happiness version 3.3.  It may lack some features you've grown to love, but, well, there you go.  It's up to you, but if you decide to go that way, just enter:
AT&U=3.3
(Assuming you compiled the source with the right amount of SPIFFS...)

Having said ALL THAT -- thanks for the bug report!  It might have been some really bad luck, or it might be a bug in 3.4 and the CONFIGURE app.  Either way, I'll investigate.  And please, ever so please, let me know if you find any other problems with those little BASIC bastards. :)

allenhuffman, 2018-02-19:
There could be something else going on. My modules often will not reconnect to my configured access point on startup. Trying to ATW"xxx,yyy" again will also repeatedly fail. I have found that if I connect to my Guest network, then connect back to the main point, it has so-far always connected that way. Petascii is not being used in my situation.

SCUBA82, 2018-02-19:
Wow that was superfast ;-) thank you for your answer!! I'm gonna trie it today

SCUBA82, 2018-02-20:
Yesterday i've successfully downgraded to v3.3 but unfortunatley with no better luck. What I've discoverd so far in this version: 

- SSID names, with spaces in it, get cut off and I can't connect to them
- The "CONFIGURE" tool has the same "upper-lower case issue" as in v3.4 (SSID shows as "aBC2") but this seems to be a display problem only.
- Numeric  only Wifi passwords work fine (also in v3.4)
- To Wifis with with alphabetic passwords I can't connect
- In CCGMS I was able to connect with the ATW command immediately. 

my guess is that something is messing up the upper-lower case syntax of the given password. 
For now I've switched back to 3.4 and saved my credentials. Everthing seems to work over several restarts. 
I think we can close this issue. it might be my environment?.

bozimmerman, 2018-03-01:
Hey Scuba82 -- I just checked in a new version of the configure app... ver 1.5.  You can get it here, or from my ftp site.  Would you mind trying it out for me?

bozimmerman, 2018-04-30:
I just checked in yet another new version of the configure app... ver 1.6. You can get it here, or from my ftp site. Would you mind trying it out for me?

bozimmerman, 2018-05-23:
Going to close this, it looks all good now.

closed 2018-05-23 by bozimmerman