bozimmerman/Zimodem #19: Build errors

drencorxeen, 2018-05-22:
I am getting some build errors that weren't happening two months ago.
`In file included from E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\zimodem.ino:132:0:

C:\Users\david\AppData\Local\Temp\arduino_build_547834\sketch\proto_zmodem.h: In function 'boolean zDownload(fs::FS&, String, String&)':

proto_zmodem.h:453: error: no matching function for call to 'fs::FS::open(String&)'

   files[0] = fs.open(filePath);

                              ^

C:\Users\david\AppData\Local\Temp\arduino_build_547834\sketch\proto_zmodem.h:453:30: note: candidates are:

In file included from C:\Users\david\AppData\Local\Temp\arduino_build_547834\sketch\pet2asc.h:30:0,

                 from E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\zimodem.ino:119:

E:\Utils\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/FS.h:113:10: note: fs::File fs::FS::open(const char*, const char*)

     File open(const char* path, const char* mode);

          ^

E:\Utils\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/FS.h:113:10: note:   candidate expects 2 arguments, 1 provided

E:\Utils\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/FS.h:114:10: note: fs::File fs::FS::open(const String&, const char*)

     File open(const String& path, const char* mode);

          ^

E:\Utils\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/FS.h:114:10: note:   candidate expects 2 arguments, 1 provided

E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\proto_zmodem.ino: In member function 'bool ZModem::setup_for_next_file()':

proto_zmodem:353: error: 'errno' was not declared in this scope

             upload_file_list[upload_file_list_i].name(), strerror(errno),

                                                                   ^

E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\proto_zmodem.ino: In member function 'void ZModem::reset_timer()':

proto_zmodem:407: error: 'time' was not declared in this scope

   time(&timeout_begin);

                      ^

E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\proto_zmodem.ino: In member function 'bool ZModem::check_timeout()':

proto_zmodem:418: error: 'time' was not declared in this scope

   time(&now);

            ^

E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\proto_zmodem.ino: In member function 'void ZModem::hexify_string(const unsigned char*, unsigned int, unsigned char*, unsigned int)':

proto_zmodem:476: error: 'assert' was not declared in this scope

   assert(output_max >= input_n * 2);

                                   ^

E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\proto_zmodem.ino: In member function 'bool ZModem::dehexify_string(const unsigned char*, unsigned int, unsigned char*, unsigned int)':

proto_zmodem:503: error: 'assert' was not declared in this scope

   assert(output_max >= input_n / 2);
[... 195 lines cut ...]
E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\proto_zmodem.ino: In member function 'bool ZModem::zmodem_process(unsigned char*, unsigned int, unsigned char*, unsigned int*, unsigned int)':

proto_zmodem:4986: error: 'assert' was not declared in this scope

   assert(input_n >= 0);

                      ^

E:\Utils\arduino-1.8.5\portable\sketchbook\zimodem\proto_zmodem.ino: In member function 'bool ZModem::zmodem_start(fs::File*, const char*, bool, ZMODEM_FLAVOR)':

proto_zmodem:5064: error: 'assert' was not declared in this scope

     assert(file_list != NULL);

                             ^

proto_zmodem:5066: error: 'assert' was not declared in this scope

     assert(file_list == NULL);

                             ^

Using library ESP8266WiFi at version 1.0 in folder: E:\Utils\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi 
exit status 1
no matching function for call to 'fs::FS::open(String&)'`

mrgw454, 2018-05-22:
I contacted Bo directly about this compile issue a few days ago in email.  He was very quick to respond and is looking into it.

drencorxeen, 2018-05-23:
That is good to know.
I am using the current stable ESP8266 and Arduino IDE 1.8.5

So hopefully that helps.

bozimmerman, 2018-05-23:
You should be all good now -- sorry about that.  Hopefully in a few weeks I'll have zmodem actually working and we can find something interesting to do with it.. in the meantime, I apologize for breaking your builds.

closed 2018-05-23 by bozimmerman