bozimmerman/Zimodem #24: 'LogMode' does not name a type

stimpy23, 2018-11-06:
Pretty sure it's not a bug but me ignoring some essentials, I just don't know about, since I'm still pretty new to the whole arduino world...
...But when trying to compile your source, I always get this Error:

`'LogMode' does not name a type`

Please find the debug output attached...
[zimodem.log](https://github.com/bozimmerman/Zimodem/files/2552286/zimodem.log)

I did follow your instructions downloading and installing/injecting the head version of the esp8266 libs and I followed instructions, I found on the internet stating I could compile using the default version from the board manager - both presenting the same error.

Please help! :)

bozimmerman, 2018-11-07:
Wow, that's a pretty fundamental problem to have.  "LogMode" is defined in "filelog.h" which is like the second or third include at the very very top of zimodem.ino, so it should be one of the first things the compiler learns about, instead of something it has never heard of.

Sending the log file is helpful.  But I don't see any missing include file errors, and your -Ddefines all look ok for what you are trying to build for.   As for differences, you're using a more recent version of the libraries, and lwip2 instead of lwip, and you are pre-compiling a bunch of libraries that mine doesn't.  

But that's the thing -- this is a pretty fundamental low-level problem -- namely not reading an existing header file, and giving no reason why.  Can you move that include to the very very top of zimodem.ino and recompiling?  I'm wondering what happens if you double-down on it (also, if there is some #define that is hiding it, this would fix that.

stimpy23, 2018-11-07:
Hey! Thanks for looking into this!
Here's the new debug log with the include of filelog.h moved all the way up. This time, he's complaining about File not naming a type:
`'File' does not name a type`

[zimodem-filelogtop.log](https://github.com/bozimmerman/Zimodem/files/2556697/zimodem-filelogtop.log)

And these are lines 14-18 of my modified zimodem.ino, so you can check, I did everything correct:
```
   limitations under the License.
*/
#include "filelog.h"
//#define TCP_SND_BUF                     4 * TCP_MSS
#define ZIMODEM_VERSION "3.4"
```

What I will do for now is completely uninstalling everything arduino and reinstalling from scratch. I wouldn't bet on me not having f*cked up anything before... ;)

I'll report back when this did the trick and if I don't, the problem remains...

stimpy23, 2018-11-08:
I just tried compiling on my Macbook (MacOS Mojave, v10.14):

1. Downloaded and installed the IDE v1.8.7 from https://www.arduino.cc/en/Main/Software
2. Added http://arduino.esp8266.com/stable/package_esp8266com_index.json to the boardmanager
3. Installed the ESP8266 2.4.2 Community libs via boardmanager
4. Replaced the installed libraries with the ones from https://github.com/esp8266/Arduino (I with and without this step - can you tell if this is still needed?)
5. Set Board to "NodeMCU 1.0 (ESP-12E Module" and the Flashsize to 4M (3M SPIFFS), left all the rest at default.
5. Hit "Verify" to compile
6. Same error as always...

These are the same steps, I did on my windows machine, except on the macbook it's a guaranteed fresh install. I never had anything arduino installed on the macbook, so theres no way any old config is causing this.
BTW: It does compile the examples (e.g. the ESP8266 WifiWebServer example) just fine..

johngeffe, 2018-11-10:
I was able to compile zimodem under all the Arduino version up-to version 1.8.5.
Version 1.8.6 seems to be the version where the error started.

stimpy23, 2018-11-10:
Guess what: ME TOO! \o/
Thanks a lot for your help! I would've never come up with the idea of trying older versions of the IDE by myself..

johngeffe, 2018-11-10:
Launching 1.8.6 from the console with this command logs the output to a txt file.
(Your path is probably different!)
`C:\Arduino\arduino-1.8.6>**arduino_debug.exe > log-1.8.6.txt 2>&1**`
Running the same command from 1.8.5 gives us two files to compare.  Not sure what we are looking for, but there seems to have been significant changes from 1.8.5 to 1.8.6 looking at https://github.com/arduino/Arduino
It might be time to put an issue to the [Arduino](https://github.com/arduino/Arduino/issues) group linking to this issue?

bozimmerman, 2018-11-11:
Huh.  Thanks johngeffe!  I'll add this note to the build instructions in the README.  Hell someday I may even try to figure out why that happens... :)

closed 2018-11-11 by bozimmerman

hpingel, 2019-07-30:
First of all: Thank you Bo for this great firmware and thank you to maintaining the sources in public!

I have compiled the Zimodem firmware yesterday for my NodeMCU (v3 - I think the v3 is hopefully irrelevant for the following).

I wanted to report that I was able to use the currently latest Arduino IDE version 1.8.9 (on Ubuntu Linux 18.04). It seems to me that whatever issue occured when using Arduino IDE version 1.8.6 has now been resolved. I can't put my finger on it when or what was resolved but 1.8.9 does compile the firmware.

Instead of the development version of https://github.com/esp8266/Arduino I had a look at the recent releases of that project and there was a recent version 2.5.2 from spring 2019 that seemed to fresh enough (compared to 2018 master branch). So I threw the following URL into my Arduino IDE to be able to consume version 2.5.2:

https://github.com/esp8266/Arduino/releases/download/2.5.2/package_esp8266com_index.json

Using this I was able to build the firmware without issues. I choose the target device "Generic ESP8266 Module" (not sure if that was the best choice but it works).

I deployed the firmware using esptool because I have issues with firmware deployment inside of the Arduino IDE.

Note: The Zimodem firmware reports a version of 3.5.0 where the README here in the repo is still at 3.4.x.

I'm very happy now! :)

bozimmerman, 2019-07-30:
I'm glad to hear things are working for you.