bozimmerman/Zimodem #25: Exception when enabling host mode

stimpy23, 2018-11-12:
More precisely: When saving after having enabled host mode.
First, it takes ages, then it throws an exception and sometimes reboots, sometimes hangs.

[exception.log](https://github.com/bozimmerman/Zimodem/files/2572138/exception.log)

This started with f801cc70fa1a5e6d60d06303e4fb7aeeddee1a60
Last known good: 7d19a21f52a4dd4205dcad5beb71ab1d1e088b3c

bozimmerman, 2018-11-21:
Hello.  I  was able to recreate this, but haven't found the fix yet -- although it seems to me that you did a lot of the work for me by finding the build where things went wrong.  So thanks for that!

In the meantime, it seems that the old commands for setting up a listener .. aka ATA6502S41=1 seem to still work ok, so you can use those.

bozimmerman, 2018-11-24:
Fix was just checked in.  Thanks again for the report!

closed 2018-11-24 by bozimmerman

stimpy23, 2018-11-29:
Hope, this doesn't reopen the issue.. ;)
Just wanted to say thank you for the workaround and for fixing this. I'll wait for your commit of the fix before trying again..