bozimmerman/Zimodem #30: Issues with Zimodem on wemos D1 mini Rev 2

rarrr, 2019-01-15:
Using Arduino 1.8.5 to build.
I'm able to build and upload the sketch to the board.
The serial monitor correctly displays the init message however most of the time it ignores any commands given.  I've noticed if I connect to to using tera term the cursor flashes when I'm able to give it commands.

I've tried two different D1 mini's with the same results.

Shorting the tx and rx pins and uploading a simple sketch shows that it's transmitting and receiving data correctly.

Seems like sometimes after showing the init message it's freezing.

rarrr, 2019-01-16:
So after playing around for a bit I've discovered if I have rx -> tx on the d1 mini, plug in the usb and then unplug the tx after ~5 seconds it works EVERY time.

bozimmerman, 2019-01-16:
I'm sorry man, but I have no idea how the 8266 chip is wired up there.  The Zimodem firmware uses more pins than just rx and tx, so I can imagine all kinds of problems that might occur if it is using the other pins in incompatible ways.

rarrr, 2019-01-16:
Fair enough.

closed 2019-01-16 by rarrr