bozimmerman/Zimodem #33: Stack-error dump with comma/exclamation mark in wifi SSID or keyphrase?

soydemadrid, 2019-02-28:
Hi, I hope you can help!

Is there anyway to recover a nodemcu unit from a massive stack error and continuous dump as this screenshot:

[image]

Everything works great accessing a strangely name wifi access point called "Fritz!Box 6490 Cable (AGL,n)"

Possibly it doesn't like the comma or the Brackets - or even the keyphrase has an exclamation mark in it!

But then when a AT&W or AT+config save takes place it just stack overflow dumps out that like and won't stop. Unplugging doesn't help as on plugging it in again the stack dump just continues...

Thanks for any help. I can get the wifi name and codes changed if I can recover my nodemcu unit somehow... But it would be even better if I could recover it AND keep the credentials as of course in future it would be useful to not have it crash if it does't like a particular access point.

Thanks once again :)

bozimmerman, 2019-02-28:
I'll look for the crash condition, but for workarounds: are you sure you're putting everything in quotes with the ATW command?

soydemadrid, 2019-03-01:
Hi thanks well currently I can't type or do anything as the stack just repeats and repeats. Is there anyway to get the modem back working? Or is it just bricked now.. Thanks again for any help

bozimmerman, 2019-11-16:
I was never able to reproduce this, nor has it been reported by anyone else.  If you see this on a second hardware unit, and maybe are able to send one to me, please re-port the issue.

closed 2019-11-16 by bozimmerman