bozimmerman/Zimodem #39: Cannot compile for ESP32

bw21p25, 2019-09-25:
I'm not able to compile this for the ESP32. I've followed directions in the read me to add the lines to the base files and tried Arduino IDE's from the version suggested to current and get compile errors.

`Arduino: 1.8.10 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 40MHz, 4MB (32Mb), 921600, None"

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1541:27: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

       lprintf(zm,LOG_INFO,"Resuming transfer from offset: %"PRIu64, *pos);

                           ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1576:24: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

     lprintf(zm,LOG_ERR,"ERROR %d seeking to file offset %"PRIu64

                        ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1659:28: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

       lprintf(zm,LOG_DEBUG,"send_from: end of file (or read error) reached at offset: %"PRId64, zm->current_file_pos);

                            ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1832:20: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

   sprintf((char*)p,"%"PRId64" %"PRIoMAX" 0 0 %u %"PRId64" 0"

                    ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1832:29: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

   sprintf((char*)p,"%"PRId64" %"PRIoMAX" 0 0 %u %"PRId64" 0"

                             ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1832:40: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

   sprintf((char*)p,"%"PRId64" %"PRIoMAX" 0 0 %u %"PRId64" 0"

                                        ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1911:24: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

   lprintf(zm,LOG_DEBUG,"Sending %s from offset %"PRIu64, fname, pos);

                        ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1927:30: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

       lprintf(zm,LOG_WARNING,"File skipped by receiver at offset: %"PRIu64, pos + sent_bytes);

                              ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1946:24: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

     lprintf(zm,LOG_ERR,"Received %s at offset: %"PRId64, chr(type), zm->current_file_pos);

                        ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:1995:25: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

     lprintf(zm,LOG_INFO,"Downloading %s (%"PRId64" KBytes) via Zmodem", zm->current_file_name, kbytes);

                         ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:2006:32: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

         lprintf(zm,LOG_WARNING,"%s already exists (%"PRId64" bytes)",fpath,l);

                                ^

[... 146 lines cut ...]
                ^

Multiple libraries were found for "WiFiClientSecure.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFiClientSecure
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\WiFi
 Not used: C:\Program
Multiple libraries were found for "SPIFFS.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SPIFFS
Multiple libraries were found for "FS.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\FS
Multiple libraries were found for "Update.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\Update
Multiple libraries were found for "SD.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SD
 Not used: C:\Program
Multiple libraries were found for "SPI.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\libraries\SPI
exit status 1
'void changeBaudRate(int)' was declared 'extern' and later 'static' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

bozimmerman, 2019-09-26:
Thanks for letting me know.  Some of this might be the compiler used.  Did you download the same version of Arduino that I mention in the README?

Either way, some of this can be done in different/older ways that I wouldn't mind switching to, but some might be due to updates in the ESP32 libraries that I need to look into.

Thanks again -- I'll keep this open until all these issues are addressed, though I might ask you to test the fixes, since I cannot.

bw21p25, 2019-09-26:
Yes I tried the version you referenced. I also tried some the newer version that was mentioned should work and the current.

bw21p25, 2019-09-27:
Also, was able to try a ESP8266 module and it does compile using current Arduino IDE 1.8.10 and ESP8266 2.5.2 library. I'm a total amateur, but if I find out anything I will update here.

bozimmerman, 2019-10-02:
So, I checked in a version that replaces all those inttypes.h strings with standard printf strings.  You might want to try the esp32 again.

bw21p25, 2019-10-04:
I tried that version, after figuring out an issue I had injected by not editing the library files properly this is what I'm left with. It still doesn't compile but it looks much closer to doing so. Thanks for your work on this.

`Arduino: 1.8.10 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\Pete\Downloads\Zimodem-master\zimodem\rt_clock.ino:17:0: warning: "htonl" redefined

 #define htonl(x) ( ((x)<<24 & 0xFF000000UL) | \

 ^

In file included from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip/lwip/ip_addr.h:41:0,

                 from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/tcpip_adapter/tcpip_adapter.h:45,

                 from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp32/esp_event_legacy.h:23,

                 from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp_event/esp_event.h:26,

                 from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp32/esp_event_loop.h:22,

                 from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiGeneric.h:27,

                 from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiSTA.h:28,

                 from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFi.h:32,

                 from sketch\pet2asc.h:17,

                 from C:\Users\Pete\Downloads\Zimodem-master\zimodem\zimodem.ino:125:

C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip/lwip/def.h:110:0: note: this is the location of the previous definition

 #define htonl(x) lwip_htonl(x)

 ^

In file included from C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:154:0,

                 from sketch\zimodem.ino.cpp:1:

C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/HardwareSerial.h:72:9: error: 'size_t HardwareSerial::availableForWrite()' cannot be overloaded

  size_t availableForWrite();

         ^

C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/HardwareSerial.h:63:9: error: with 'int HardwareSerial::availableForWrite()'

     int availableForWrite(void);

         ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_ftp.ino: In function 'bool doFTPLS(ZSerial*, const char*, int, const char*, const char*, const char*, bool)':

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_ftp.ino:355:31: warning: NULL used in arithmetic [-Wpointer-arith]

   if((req != NULL)&& (*req != NULL))

                               ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_http.ino: In function 'WiFiClient* doWebGetStream(const char*, int, const char*, bool, uint32_t*)':

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_http.ino:210:12: warning: converting 'false' to pointer type 'WiFiClient*' [-Wconversion-null]

     return false;

            ^

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino: In function 'int zmodem_send_hex(zmodem_t*, uint8_t)':

C:\Users\Pete\Downloads\Zimodem-master\zimodem\proto_zmodem.ino:363:16: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

   char* xdigit="0123456789abcdef";

                ^

Multiple libraries were found for "Update.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Update
Multiple libraries were found for "SD.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program
Multiple libraries were found for "SPI.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
Multiple libraries were found for "WiFiClientSecure.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFiClientSecure
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program
Multiple libraries were found for "SPIFFS.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS
Multiple libraries were found for "FS.h"
 Used: C:\Users\Pete\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

bozimmerman, 2019-10-28:
I downloaded the latest ESP32 library from the url in the readme, as of today (about an hour ago), made the changes as listed in the >updated< README, and everything looks good now.

It appears they added their own writesRemaining method, but everything was otherwise OK.

bozimmerman, 2019-10-28:
... Well, it compiled ok ... but its crashing at boot time now. :/

bozimmerman, 2019-11-16:
This is definitely and completely fixed now.

closed 2019-11-16 by bozimmerman