bozimmerman/Zimodem #41: Wipe all phonebook

FujiMan66, 2020-01-22:
Command to wipe phonebook. I have an entry that wont delete.

bozimmerman, 2020-01-22:
You can reset the modem to factory settings using at&f .. that will delete and reset the phonebook, listener configuration, and standard config.

closed 2020-01-22 by bozimmerman

FujiMan66, 2020-01-25:
Thank you. I'm trying to download some files from a telnet bbs which
requires telnet mode. I've tried ATDT, but the transfers timeout.

I'm using an Atari ST 19.2k zimodem 3.4 esp8266 connecting to
darkforce-bbs.dyndns.org:520 using Y-modem batch or Z-modem.

Are there other AT commands that I should be using? ATS46-5X?

Thanks
Rory McMahon

bozimmerman, 2020-01-28:
Hello Rory.
You definitely don't want to use firmware TELNET mode when connecting to a "telnet" BBS for file transfers.  This is because the file transfer protocols (xmodem, zmodem, etc) will "incidentally" send telnet escape codes during the normal coarse of transferring binary data, which will get captured by the firmware before it gets to your computer.  

Most online BBSes call themselves "Telnet BBS"es, not because they use the special telnet codes, but just to distinguish them from "Phone" type BBSes.  You should be perfectly safe to use ATD instead of ATDT.