bozimmerman/Zimodem #42: Suggestion: Enhanced phonebook and ping functionality

mczero80, 2020-04-12:
I don't know if it is ok to suggest features under the 'issues' tab, but those are some features that would be nice:

1. An enhanced phonebook with BBS names. Often I don't know which hostname belongs to which bbs.
2. Ping functionality. To check if a IP or hostname responds. I don't know if that works, but automatic pinging all phonebook entries to check which BBSs are online would be nice, too! 

I love Zimodem. Really nice to have it! Thank you.

bozimmerman, 2020-04-12:
It's perfectly fine!

bozimmerman, 2021-03-16:
a 'notes' field has been added to the phonebook in the latest src.

bozimmerman, 2023-03-05:
Ping added.  This is all done now.

closed 2023-03-05 by bozimmerman

Biker1bob, 2023-03-05:
This is great stuff Bo, thanks so much.