bozimmerman/Zimodem #43: Compile errors on ESP-32 NodeMCU board

baldrick1, 2020-05-02:
Getting the following error when attempting a compile using ESP32 Dev Module:

C:\Users\can77\Downloads\Zimodem-master\Zimodem-master\zimodem\zimodem.ino: In function 'void changeBaudRate(int)':

zimodem:291:17: error: expected unqualified-id before '.' token

   HardwareSerial.changeBaudRate(baudRate);

                 ^

C:\Users\can77\Downloads\Zimodem-master\Zimodem-master\zimodem\zimodem.ino: In function 'void changeSerialConfig(uint32_t)':

zimodem:305:12: error: 'class HardwareSerial' has no member named 'changeConfig'

   HWSerial.changeConfig(conf);

dittman1, 2020-07-27:
I have the same issue.

bozimmerman, 2020-07-27:
I'm guessing the instructions in the README were not followed?

dittman1, 2020-07-27:
I don't think I missed anything.  I should have included more details and the full errors.  I'm building on a Mac and I'm using Arduino 1.8.13.  I've changed the files listed for the ESP32.  These are all the errors:

zimodem:286:40: error: 'void changeBaudRate(int)' was declared 'extern' and later 'static' [-fpermissive]
 static void changeBaudRate(int baudRate)
                                        ^
In file included from /Users/dittman/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Arduino.h:154:0,
                 from sketch/zimodem.ino.cpp:1:
/Users/dittman/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/HardwareSerial.h:106:6: note: previous declaration of 'void changeBaudRate(int)'
 void changeBaudRate(int baudRate);
      ^
/Users/dittman/Desktop/github/Zimodem/zimodem/zimodem.ino: In function 'void changeBaudRate(int)':
zimodem:295:12: error: 'class HardwareSerial' has no member named 'changeBaudRate'
   HWSerial.changeBaudRate(baudRate);
            ^
/Users/dittman/Desktop/github/Zimodem/zimodem/zimodem.ino: In function 'void changeSerialConfig(uint32_t)':
zimodem:309:12: error: 'class HardwareSerial' has no member named 'changeConfig'
   HWSerial.changeConfig(conf);
            ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/dittman/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /Users/dittman/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
'void changeBaudRate(int)' was declared 'extern' and later 'static' [-fpermissive]

baldrick1, 2020-07-27:
I have a similar situation.  Followed the instructions in the readme file as best I could.
Perhaps there’s a setting we missed.

Henry

dittman1, 2020-07-28:
I was able to get it to build but it doesn't work.  I removed the Arduino environment completely along with all the libraries and source files, then reinstalled the Arduino IDE and the ESP32 core from github (stable release) as detailed here:

https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

I then applied the changes listed.  The only difference from before is where I put the lines in HardwareSerial.h.

After that I could successfully build and upload.  However on reset all I get on the serial monitor is output at 115200 of:

Reset complete.  Init start
Init complete.

At 1200 baud I get nothing.

djchez01, 2021-02-08:
Was there ever any resolution to these issues? I have run across the same and am wondering if anyone figured it out.

bozimmerman, 2021-02-09:
So, are the compile errors identical?
My only guess is that maybe they changed something about the compiler along the way?  I'm using Arduino IDE 1.6.11 still..

djchez01, 2021-02-09:
Bo, thanks for responding: I used the version from your notes: "To build this firmware for the ESP-32, I also used the Arduino IDE rev 1.8.10 with the Arduino ESP32 Dev Module installed from the board manager" - on windows 10. I keep getting the  'class HardwareSerial' has no member named 'changebaudrate'. I did try updating to the latest version but using your changes in the notes, but got a ton of errors. I need to revert back to 1.8.10.

The device I have are labeled: NodeMCU ESP32-S and I got them off Amazon.

Thanks for making this firmware - I use it on the ESP8266, and it is great!

bozimmerman, 2021-02-12:
Can you double-check HardwareSerial.h has the void changeBaudRate(int baudRate);.
If it does, then it must not be looking at the same HardwareSerial.h you think it is.  In which case, check the arduino ide docs on how to turn on compiling command line echo.  In mine it's under File/Preferences-> "Show verbose output during [X] Compilation".  

Then, when compiling, you'll see lines like this:
C:\Users\Bo Zimmerman\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\Bo Zimmerman\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\Bo Zimmerman\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid"  ...........

The trick is to look and see if the directories on the -I bits matches where you found HardwareSerial.h.  You may have to copy/paste a line into notepad and word on word wrap to poke through it.  

Let me know!

bozimmerman, 2021-03-15:
Closing this for lack of communication (and not being able to repro).  LMK if something changes.

closed 2021-03-15 by bozimmerman