bozimmerman/Zimodem #47: Carrier Detect line not being activated on manual answer ESP8266. (Again)

dabonetn, 2020-08-14:
Ok, I got around to testing your new version after the pull, but cd is still not activated on connect in the following scenario.

Device is a ESP8266 on a wemos d1-mini clone board.
Serial side is connected to a full (10 wire), ttl to serial adapter board. This is attached to my quad port rs232 card on a windows 10 system.

The only change to the source is commenting out line 78 //# define RS232_INVERTED 1 because it's a non inverted device.

Flashed the unit with the latest pull as of this post time, joined my wifi network, enabled cts/rts flow control, on the main menu and in the bbs host menu.

Saved these settings, and exited back to command mode.

Called the zimodem using putty with a raw port. This results in a connection and the Carrier Detect line becoming active and the led on the wifi module to light, along with the led I've added to my breadboard.

Just closing putty results in a NO CARRIER (what ever number) and the LEDs going out.

Now, type ats0=0, this disables autoanswer on incoming connections, and results in the modem reporting RING.

Starting the putty connection again, results in a RING on the serial terminal side, and when I make the connection with the ATA command, I am connected, but no LEDs, i.e. DCD is not being changed to reflect the current connected state.

dabonetn, 2020-12-29:
Ok, current testing with 3.5.6

I enable BBS mode, and connect with auto answer, dcd activates (goes low).
When I disable auto answer, and connect using the ata command after a ring, dcd doesn't activate, unless I drop to command mode using +++.

After dropping to command mode, dcd immediately activates, and issuing ato resumes the connection and the dcd line works correctly for that session (i.e. drops when the connection drops)

bozimmerman, 2021-03-13:
This is a fantasticly detailed reproduction story.  I have not had luck reproing this in the past, but I will print out your steps and follow them closely on my next try.

bozimmerman, 2021-03-15:
OK, this should be fixed now, thanks again!

closed 2021-03-15 by bozimmerman