bozimmerman/Zimodem #53: No output on serial console

0C-34, 2020-12-31:
Hi,
I'm trying to get zimodem to work on a esp32 (i'm trying both a TTGO-TS and a normal ESP32 dev board with a ESP32-WROOM-32 module soldered on it) but on the serial monitor i get only the debug info, if switch to 1200 baud and write "AT" or "AT+CONFIG" i get nothing.

I am using arduino studio 1.8.10 in "portable mode" (so I don't mix up the libraries between projects) with the 1.0.0 libraries that i found linked in the readme (I didn't make the changes described because i noted that they were already applied).
I also tried modifying the 1.0.4 library appending all the changes at the end of the files (have no idea if it's "the right way) but with no luck.
<details>
  <summary>compilation log</summary>
    
zimodem:284:40: error: 'void changeBaudRate(int)' was declared 'extern' and later 'static' [-fpermissive]

 static void changeBaudRate(int baudRate)

                                        ^

In file included from sketch\pet2asc.h:19:0,

                 from C:\Users\Sonno\Documents\arduino-zimodem\portable\sketchbook\zimodem\zimodem.ino:125:

C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32/HardwareSerial.h:107:6: note: previous declaration of 'void changeBaudRate(int)'

 void changeBaudRate(int baudRate);

      ^

C:\Users\Sonno\Documents\arduino-zimodem\portable\sketchbook\zimodem\zimodem.ino: In function 'void changeBaudRate(int)':

zimodem:293:12: error: 'class HardwareSerial' has no member named 'changeBaudRate'

   HWSerial.changeBaudRate(baudRate);

            ^

C:\Users\Sonno\Documents\arduino-zimodem\portable\sketchbook\zimodem\zimodem.ino: In function 'void changeSerialConfig(uint32_t)':

zimodem:307:12: error: 'class HardwareSerial' has no member named 'changeConfig'

   HWSerial.changeConfig(conf);

            ^

Multiple libraries were found for "Update.h"
 Used: C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\libraries\Update
Multiple libraries were found for "SD.h"
 Used: C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Users\Sonno\Documents\arduino-zimodem\libraries\SD
Multiple libraries were found for "SPI.h"
 Used: C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
Multiple libraries were found for "WiFiClientSecure.h"
 Used: C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFiClientSecure
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Users\Sonno\Documents\arduino-zimodem\libraries\WiFi
Multiple libraries were found for "SPIFFS.h"
 Used: C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS
Multiple libraries were found for "FS.h"
 Used: C:\Users\Sonno\Documents\arduino-zimodem\portable\packages\esp32\hardware\esp32\1.0.4\libraries\FS
exit status 1
'void changeBaudRate(int)' was declared 'extern' and later 'static' [-fpermissive]
</details>

also i noted that in the HardwareSerial.cpp there's no
```_uart = uartBegin(_uart_nr, baud, config, rxPin, txPin, 256, invert);```
but there is something similar
```_uart = uartBegin(_uart_nr, baud ? baud : 9600, config, rxPin, txPin, 256, invert);```

Hope to be able to navigate on my i386 by 2021, have a happy new year :P

bozimmerman, 2021-03-13:
*ALREADY APPLIED* ?! That's awesome!

However, your having troubles is not awesome.  
My first question is whether anything happens when you type AT and press enter.  Do you get the initialize message with the board info when you reset?

0C-34, 2021-03-14:
> _ALREADY APPLIED_ ?! That's awesome!
They're literally the libraries you use in production, or at least that what's written inn the Readme lol

BTW yes, when I restart the board I get the usual board info at ~11kbaud on the arduino serial monitor

From the old computer side I can't write anything, on the arduino monitor side I can write but I get no response, I'll double check tomorrow just to be sure (it's 2AM here rn 😂)

Thanks for the reply have a good night/day!

bozimmerman, 2021-03-14:
Hi -- I couldn't quite tell from your answer:  you see the startup message on the old computer side I assume: it should end with INITALIZED and then READY, right?  Is that correct?
I understand you are not able to type from the old computer side.  What are your old computer flow control settings?
The debug serial side (115kbaud) isn't interactive -- it just sends debug messages. So that's not supposed to let you type anything.

0C-34, 2021-03-14:
I just tried now (last time I tested this was a month ago) and this is the result:
<details>
<summary>Arduino monitor side, 115200 baud</summary>
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
Debug port open and ready.
Baud change to 1200.
Deque constant now: 4
Config changing 134217756.
Deque constant now: 4
Config changed.
Reset start.
Reset complete.  Init start
Init complete.
</details>

<details>
<summary>Putty side, 1200 baud</summary>
v▒▒▒▒▒▒▒
</details>
Also the LED on the TX line of the MAX3232 blinks while i get the garbage characters.
I've checked and if i set Putty on 115200 the garbage characters are actually the same message sent to the arduino serial monitor.

If I send a message trough the arduino monitor i get no response.
If I write on the putty shell i get the led on the RX line to blink, but there are no characters drawn on the shell.
If I type blindly "AT" and press "enter" i get no activity on the TX line and nothing on the screen.

bozimmerman, 2021-03-15:
Wow, so, hmm.   It seems like UART 1 and UART 2 in my firmware might be mapping to the same physical serial port on your hardware.  We may be looking at some incompatibility between your unit and the ESP32 libraries I use -- since I rely on those to keep this stuff straight.

0C-34, 2021-03-15:
I tried 2 different boards: one has a esp32-wroom-32 module soldered on it and the other is this thing [https://github.com/LilyGO/TTGO-TS](url)
One thing in common is the presence of the CP210x USB to UART bridge.
They both have the same problem as described before.
Also I have updated the project to the latest one on the repo as of yesterday and the problem has not changed.
Thanks for the libraries, but deleting the folder of the old ones and putting these newer ones didn't change the problem (at least it didn't break anything ':D)

I am using arduino in portable mode, so it doesn't mess up any other projects i need to compile and so I don't have problems while compiling with either esp8266 and esp32, I just need to keep the projects in different folders each with their own copy of arduino and libraries.
This means that I can zip the folder and directly send it to you if you prefer, so we can rule out any human error on the software side.

bozimmerman, 2021-03-15:
Well, in that case, all that's left is to get ahold of similar hardware.  That TTGO-TS seems way overkill ..  if you can give me a grocery list and some instructions on your esp32 board, I'll give it a shot.  I'm afraid I'm not a hardware guy, so giving me instructions like "get something sorta like X and then just hook it up to Y" will go absolutely nowhere with my brain. :( I can follow simple instructions though, and I own a soldering iron!

0C-34, 2021-03-15:
I just looked up the pinout for the generic devboard and I've seen that there are "rx0" and "tx0" pins (where I was connected) and "rx2" and "tx2" that I totally didn't see on the silkscreen
Sure enough connecting the max3232 to those pins makes it work...
It's beyond me why the second board has only the UART0 pins exposed but whatever, would've been cool to send some info to the screen and use the built-sd adapter (and also maybe make the speaker do some tones :B).

Still, I am really sorry for wasting your time.
If you want/have the time please update the readme to incorporate those suggestions:
A) downloading a fresh copy of arduino, creating a folder called "portable" in the arduino executable's folder so the changes made to the library are not system-wide
B) specify that you have to hook the max3232 (or whatever other ttl to rs232 conversion chip) to the UART2 port on the ESP side

I have now to check the code to see if it's possible to swap the 2 UART ports.
Have a good evening!

closed 2021-03-15 by 0C-34