bozimmerman/Zimodem #55: Feature Request: HTTPS request support

flimshaw, 2021-02-25:
So, I have no idea if this is possible to implement on an ESP8266 (there seems to be [some work on the subject](https://maakbaas.com/esp8266-iot-framework/logs/https-requests/)), but being able to make requests/posts to an https location would really expand the sorts of things you could do with one of these on an otherwise untrusted wifi network. 

Also, thanks so much for sharing this project! It was fun and easy to knock together a modem with an old 8266 I had lying around, and it really opens up a lot of possibilities on my TRS-80 and C64.

virtadpt, 2021-02-25:
It is possible to do HTTPS on an ESP8266.  I use Micropython running on an 8266-based Adafruit Feather with the urequests module.

https://github.com/virtadpt/exocortex-halo/blob/master/environment-sensor-esp8266/main.py#L263

If it can be done in Python in a fairly lightweight manner, it should be possible to implement it in a lower level language on this platform.

bozimmerman, 2021-03-13:
The big problem with SSL in Zimodem is the amount of space required.  The firmware butts against the size limit right now, which is 1/2 available program space -- the other half is required for OTA updates.  Considering all the root key data, different protocols, etc, it can be a mess.
For user happiness, the HTTPS layer needs only to accept every key and proceed in order to cover the basics.  However, the SSL impl don't seem to do that well -- they really expect you to provide valid root keys.  There may be around this though.. it's on my TODO list, though any help would be appreciated.

bozimmerman, 2021-03-13:
That said, if the firmware is built for the ESP32, then you do get the WiFiClientSecure object whenever using an https:// url... complete with the aforementioned flaws.

closed 2021-03-14 by bozimmerman