bozimmerman/Zimodem #57: atw command doesn't work if multiple access point of the same name exists.

dabonetn, 2021-03-16:
I run a meshed wifi system, and if I try to join the network using the atw command, and not the config menu it always fails.

Doing the atw command lists all 3 access points, with the different strengths. 

My suggestion is instead of doing a blind join, parse the available ssids, see if any match the input.
If none match the wanted ssid, print a error stating Incorrect SSID or SSID no Found.

If you have multiples of the wanted ssid, select the one with the best signal, and then join it.

dabonetn, 2021-03-17:
Btw, this also prevents being able to use fixed ip currently.

bozimmerman, 2021-03-20:
It feels like having to poll the ssids on startup just adds a delay for the benefit of one unique and unusual situation.

The static ip/etc is in the current version, and works fine w/o this.

dabonetn, 2021-03-20:
Then maybe add it to the config menu? With the current command, there is no way to assign a static IP when you have a mesh network.

bozimmerman, 2021-03-20:
So, if I understand you correctly, you have numerous wifi routers, each with their own subnet, but they all have the exact same name.  Hmm...  Unfortunately, the arduino api only accepts a string as an ssid, so some hacking of the underlying libs would be required.

dabonetn, 2021-03-20:
No, I have multiple access points for the same network and subnet. They are spaced around the house to give better coverage for all the devices. (One is actually at the neighbors house using a wireless bridge to connect him to the lan).

It's basically your standard multi access point network, devices usually just attach to the access point with the best signal.

I'm using Unifi Access points, but lots of companies make this kind of setup. You'll also see this kind of setup sold as whole home wifi.

dabonetn, 2021-03-20:
Just tried again with a PC and it worked fine, please close this, apparently I was doing something wrong. 
I was trying it on my commodore, maybe I wasn't getting something right.

My bad on this one. Thanks.

closed 2021-03-21 by bozimmerman