bozimmerman/Zimodem #59: Feature Suggestion: Busy string.

dabonetn, 2021-03-18:
Is there a way to respond to another incoming connection while connected with a text string?

I.e. max listeners is 2, if listener 0 is active then 1 automatically responds with the busy message and then auto disconnects.

Like "I'm sorry the bbs has another user currently, please call back later."

bozimmerman, 2021-03-19:
Yea, technically it tried to send the word BUSY, but socket flush just doesn't work, so the other side doesn't even get it before the socket is closed.  I know what I have to do: create a que of existing connections that need to be closed at a later time.  I just need to do it.

SO! Thanks for the reminder!

dabonetn, 2021-03-19:
What about leaving it open until a character is received back, or until 5 seconds is reached?

bozimmerman, 2021-03-20:
Just checked this fix in.  Also made the msg configureable.

closed 2021-03-20 by bozimmerman