bozimmerman/Zimodem #61: 'class HardwareSerial' has no member named 'changeBaudRate'

Jason-Weber-UC, 2021-03-22:
Trying to compile zimodem on Arduino IDE 1.8.5 using (either) the latest stable or Dev Espressfi ESP32 board (ESP32-WROOM-32).
Running compile gives me 
'class HardwareSerial' has no member named 'changeBaudRate'
Board is set up exactly as per instructions in the IDE

Jason-Weber-UC, 2021-03-22:
I found your esp32 lib download on another thread.
Copied the files over the standard espressfi ones, and all I get is this:

C:\Users\Lurch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5-rc7\libraries\Update\src\HttpsOTAUpdate.cpp:16:29: fatal error: esp_http_client.h: No such file or directory

compilation terminated.

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Lurch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5-rc7\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\Lurch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5-rc7\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board ESP32 Dev Module.

Jason-Weber-UC, 2021-03-22:
Nevermind. The IDE was having a spit between your library's and the official ones.

closed 2021-03-22 by Jason-Weber-UC

bkw777, 2022-07-19:
What did you do? I have the same error.

wx4cb, 2022-07-24:
> What did you do? I have the same error.

me 2

bkw777, 2022-07-25:
Actually I got past that exact error by adding the functions to the library files as directed in the main readme, but just immediately hit other errors. The readme just says to add the lines, which is ambiguous, it doesn't say where. The code simply does not build under the current versions of the arduino libraries. I haven't tried using the exact v1.8.10 from the readme yet.