bozimmerman/Zimodem #62: esp32 build vs esp 8266

pottendo, 2021-04-25:
hi,
thanks for maintaining this modem SW and all the stuff around! Much appreciated!

Having built a cable (with level shifters - see below) I now can connect successfully via:
a) an esp8266
b) an esp32
However I do see some differences between a) and b)
a) basically works flawlessly. 
I've tried several c64 modem clients and e.g. with ccgms2017 I can use the UP9600/EZ232 modem type to connect with 9600 baud
with ccgms2021 I must switch to zimodem fw and up9600/EZ232 modem, the standard userport modem won't work

b) works somehow but
in ccgms2017 only standard userport modem seems to work (up to 2400 baud), when I switch to UP9600/EZ232 the c64 client crashes very qickly. e.g. when I hit the '"' (double quote) key on the c64.
I've tried to debug on the esp side, but as the c64 crashes (lock-up) I don't see too much. I've not started to debug the c64 side. I'm not that fluent in assembly anymore... and probably lack the right tools (freezers, etc.)
ccgms2021 doesn't seem to work at all; I can't get a connection to the esp32 modem.

I'd like to add a few more things (e.g. mqtt protocol) so the esp32 version would be more convenient (debug port, etc.).
Any idea why the esp32 build is not that good working for me as the other one?

Note: I've built with platformIO/VSCode (not witin the Arduino IDE) with and without your esp-hal patches - there doesn't seem to be any difference (to me). Let me know how you encountered the issues (corruptions as you write) to check if the newest esp-hal is sufficiently fixed (I use 1.0.6 as of today).

thanks alot for all you're doing to the retro-community!
pottendo

PS: I use this schematic for my cable:
  https://www.c64-wiki.de/wiki/Datei:esp8266save.jpg
of course adapted to esp32: 
  RX2/TX2 pins (16/17) and CTS/RTS slightly adapted:
# define DEFAULT_PIN_CTS GPIO_NUM_21 // was 13
# define DEFAULT_PIN_RTS GPIO_NUM_19 // was 15 unused
(simply for convenience on my breadboard setup, to have all connectors on one side)

pottendo, 2021-04-25:
hi again,
I found that the esp32 version works as good as the esp8266 version (for me) if I add
  \# define RS232_INVERTED 1
to the respective esp32 section in zimodem.ino.
I just used UP9600/EZ232 config in ccgms2021 with 9600 baud to surf the retro-bbs-net - cool experience!
Now my build uses the stock esp32 1.0.6 hal (uart) - so I think your patch recommendation is not necessary anymore. But better try yourself.

I can start hacking now... if you're interested in some results, let me know and we can have a chat about some plans I had.

bye, pottendo

pottendo, 2021-04-29:
hi,
after playing around more I found that some of my conclusions above were wrong. 
Both the esp32 _and_ the esp8266 struggle at 9600 baud. 
I suppose that flow-control isn't working well - maybe the cable/connector isn't right for this - see above what I used as schematic to build a connector to my C64
I found this by connecting to some BBS (e.g. 13thfloor) where some petscii graphics are shown, where one easily sees that some chars/lines are missing. with 4800 baud and below all seem fine.

On the C64 CCGMS doesn't crash with setting to UP9600/EZ232 modem - it just doesn't receive the proper signals on the RTS line - I found this by removing the wire from this pin on the esp32, when the C64 seemed to be locked; and it just continued.
So defining RS232_INVERTED somehow disable this flow-control pin, so it look like 'working'.

Lastly I found that the IRC client from the 8bit SW collection won't work with my ESP32 modem version. Interesting is that it does work with my esp8266 modem; I tried to debug but failed to find an explanation.

Any further input is much appreciated. I'd like to continue with the esp32 version (ease handling) but IRC isn't working at all and flow-control doesn't seem to do something. I now suspect that my HW wiring is different and doesn't match the requirements to drive flow-control properly - a wiring diagram (for a 9600 compatible) esp32 would definitely help... 

bye pottendo

Neukam, 2021-05-19:
Make sure you have your output states correct.  For a C64 you should have:

define DEFAULT_DCD_HIGH  LOW
define DEFAULT_DCD_LOW  HIGH
define DEFAULT_CTS_HIGH  HIGH
define DEFAULT_CTS_LOW  LOW
define DEFAULT_RTS_HIGH  HIGH
define DEFAULT_RTS_LOW  LOW

You may need to have DCD connected to pin H for some software to work correctly.  You may need to add another level shifter to protect an ESP32.  I have it connected directly with an ESP8266 with no issues, but I haven't tried an ESP32.  To verify that CTS/RTS are correct you can **set modem to UP9600** in CCGMS with flow control and echo on.  You should receive the echo as you type.  Type ATS50=1 to invert RTS (output, connected to CTS pin K on C64).  The C64 should stop transmitting to the ESP and you should stop receiving the echo.  Reset the ESP and after the initialization, you should receive the echo again.  Type ATS48=1 to invert CTS(input, connected to RTS pin D on C64).  The ESP should stop transmitting to the C64 and you should stop receiving the echo.  Reset the ESP and after the initialization, you should receive the echo again.  If those tests pass, flow control should be working properly.

pottendo, 2021-05-22:
> Make sure you have your input/output states correct. For a C64 you should have:
> 
> define DEFAULT_DCD_HIGH LOW
> define DEFAULT_DCD_LOW HIGH
> define DEFAULT_CTS_HIGH HIGH
> define DEFAULT_CTS_LOW LOW
> define DEFAULT_RTS_HIGH HIGH
> define DEFAULT_RTS_LOW LOW
> 
> Make sure your User Port connector is wired properly. I've seen some connectors labeled incorrectly. There is no "G", there is no "I" and there is no "O".
> 
> You may need to have DCD connected to pin H for some software to work correctly. You may need to add another level shifter to protect an ESP32. I have it connected directly with an ESP8266 with no issues, but I haven't tried an ESP32. To verify that CTS/RTS are correct you can **set modem to UP9600** in CCGMS with flow control and echo on. You should receive the echo as you type. Type ATS50=1 to invert RTS (output, connected to CTS pin K on C64). The C64 should stop transmitting to the ESP and you should stop receiving the echo. Reset the ESP and after the initialization, you should receive the echo again. Type ATS48=1 to invert CTS(input, connected to RTS pin D on C64). The ESP should stop transmitting to the C64 and you should stop receiving the echo. Reset the ESP and after the initialization, you should receive the echo again. If those tests pass, flow control should be working properly.

Great - thanks for the information; I've already found that the H pin from the UserPort was missing (DCD) -  Will try asap (need to re-setup my breadboard) with you proposed changes to the line handling macros.
PS: I'm waiting to receive my PCB which will feature a RPi shield with Pi1541 function and the modem connectivity to a esp32... 
bye, pottendo

Neukam, 2021-05-22:
That's awesome.  Please share your pcb project.  I'd be very interested in a multifunction device like that, especially if it has status LEDs for everything.  I'm using the svenpetersen1965 board with esp8266 and a pi0/epyx combo right now, but i think a wifi/pi combo would be more convenient.  I've thought to design one myself, but i haven't had the time.

pottendo, 2021-05-22:
> That's awesome. Please share your pcb project. I'd be very interested in a multifunction device like that, especially if it has status LEDs for everything. I'm using the svenpetersen1965 board with [...]

hi,
sure - will do; once it has arrived and assembled I'll update you if it's working...
Maybe I can share the Frizing file with you already now... not sure if this is fully self-contained.
Be warned: it's my very first PCB and I don't really know what I've done... ;-) I'm playing with esp's since quite a while and soldered things (2.54mm rasters) myself. 
I'm not at all skilled with that - as I'm far more into SW than HW. But it was fun to use the Fritzing tooling - and the production is really cheap (~15EUR for 3 boards incl. shipping!).
bye, pottendo

pottendo, 2021-05-31:
> That's awesome. Please share your pcb project.
hi Neukam,

@bozimmerman: this is already off-topic - sorry. I'll connect with @Neukam somehow and will close the subject here asap.

@Neukam: my PCB has at least 2 flaws: 
a) I've overlooked that the holes need to be checked for all parts. Some chosen part (e.g. ESP32 D1 module) from some source added to the Fritzing lib defined only very small hole diameters, which won't allow pinheaders to fit in. Same for some other connector places -> my prototype build worked around that.
b) I've missed one connector - this can be easily patched.
but most annoyingly, the chosen level-shifters just don't work as expected. I've no clue yet, what's the reason. 

Let's connnect somehow; don't know how this works on github - any suggestions?
bye, pottendo

Neukam, 2021-05-31:
@pottendo GitHub would be a great place to upload your files, even in their flawed state.  You can email me at dukeneukam-at-msn-dot-com to discuss your project directly.  I'm a noob at GitHub and pcb design, but I'm happy to share my thoughts.

pottendo, 2021-05-31:
connected.

closed 2021-05-31 by pottendo