bozimmerman/Zimodem #64: Issue on ESP8266 NODEMCU

pyr0hax, 2021-05-26:
With the latest build I am having an issue where after flashing, it will not take any commands.

[image]

dabonetn, 2021-05-27:
I had the same issue on one of mine, I just selected full erase when writing the sketch again and it worked after that.

pyr0hax, 2021-05-27:
> I had the same issue on one of mine, I just selected full erase when writing the sketch again and it worked after that.

I am going to ask you a very stupid question. How do you do a full erasure on arduino ide as I am using the ino files to upload to the esp8266? I did a quick google search but did not find anything useful.

pyr0hax, 2021-05-27:
> I had the same issue on one of mine, I just selected full erase when writing the sketch again and it worked after that.

Lol I found it. Thanks. Going to try this now and see what happens

discoball78, 2021-06-09:
I'm seeing the same issue here, using NodeMcu V3

[image]

Have tried using erase settings to no avail:

[image]

bebrown, 2021-06-11:
Same issue here. Sometimes it is Exception 0, other times Exception 28.

I have tried erasing the flash. Tried a different ESP8266 module.

Setup is: Mac OS Big Sur, Arduino 1.8.13, board manager esp8266 3.0.0, ZiModem 3.6.3

omiq, 2021-07-15:
Same, with both a commercial C64wifi adapter and my esp8266 boards, and both the latest and the specified versions of Arduino IDE. I found some older firmware that works but I would rather get this going as it is more feature-rich

```--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bf70 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000014 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 0190
3fffff70:  4020c360 3ffef968 3ffef964 3fff0c48  
3fffff80:  3fffdad0 00000000 3ffef5ec 40206304  
3fffff90:  3fffdad0 00000000 3ffef900 4020506a  
3fffffa0:  feefeffe 00000000 3fff0c34 4020ab20  
3fffffb0:  feefeffe feefeffe 3ffe8614 40100ca1  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------⸮⸮⸮
```

dabonetn, 2021-07-16:
Which ESP8266 lib is everyone using?

I'm still on 2.7.4

bozimmerman, 2021-08-15:
Again, thanks to Dabo for figuring out that the head revision of the 8266 libs is no longer safe to use.  I've updated the README with his recommended rev.

closed 2021-08-15 by bozimmerman