bozimmerman/Zimodem #65: Any benefit to connecting DCD/DTR/DSR/RI pins?

sofakng, 2021-06-09:
I'm a big fan of the NulSom Ultra Compact RS232-to-TTL converter but it doesn't provide the following pins:

Data Carrier Detect (DCD)
Data Terminal Ready (DTR)
Data Set Ready (DSR)
Ring Indicator (RI)

I know they aren't required for a single serial communication (ie. RX/TX, GND) but do they provide any benefit/features with Zimodem?

dabonetn, 2021-06-12:
For usb... not really. For connecting to real uarts... definitely.

Gabeki17, 2021-06-23:
Is there a wiring diagram for c64?

dabonetn, 2021-06-30:
Look for a wiring diagrams for flow control on the 64. The UP9600 interface spec supports flow control at 9600 baud on a 64.

Gabeki17, 2021-06-30:
I have found this: https://www.c64-wiki.de/wiki/Wifi-Adapter

bozimmerman, 2021-07-11:
In my limited experience, DTR and DSR are really only necessary for real UARTS -- such as in the Plus/4, SuperPET, or PC-Compats and the like.   RI, and DCD are necessary if you want to run a BBS, and DCD is required on the C64 for lots of my little util programs, since its a quick and easy way to check a remote connection.

closed 2021-07-11 by bozimmerman