bozimmerman/Zimodem #68: telnetd64 v1.0 not working

Gabeki17, 2021-06-24:
Hello,

After starting and seting the variables, right after init, I am receiving the message "Carrier detected. Reset Modem, wait 30 seconds, then hit enter:"

In it is endles if I hit enter.

I am almost sure that it is an "issue betwean the chair and the keyboard of my C64". But yes. I am stuck right here.

With regards,
Gábor

bozimmerman, 2021-07-11:
What version of the firmware do you have?  I haven't played with it in awhile, but will make sure it is still working fine on 3.6.3.

bozimmerman, 2021-07-15:
So, the latest firmware rev for the C64 is 3.6.1, and last night I tested telnetd on it, and had no troubles.  
Now that I think about what you're seeing, I wonder if you are using custom hardware that does not have the DCD line correctly connected to the user port?

bozimmerman, 2021-07-16:
No response received, and no reproduce, so closing.

closed 2021-07-16 by bozimmerman

Gabeki17, 2021-07-21:
Thx!

Sorry  I was away for 2 weeks and could not follow.

I have this wiring: https://www.c64-wiki.de/wiki/Datei:esp8266save.jpg

bozimmerman, 2021-07-22:
Hi.  It appears the problem is that you don't have DCD on GPIO2 hooked up to user port pin H.  The telnet server relies on that signal to know when it is connected.  
I would recommend swapping your CTS on pin K for H, and then tying pin K to 5V through a 10k resistor.

Gabeki17, 2021-07-23:
Thx!

Why not connecting D4 (GPIO2) to DCD simply (To "H")?

K is now connected to D2 (GPIO4)

I think you have a well prepared reason that I would like to understand.

Thank you very much!

I am really enjoying your responsiveness and willingness to help me!

With regards,
Gábor

Gabeki17, 2021-07-23:
Ok I did this. (D4 (GPIO2) to DCD simply (To "H"))

And now Works! Thx!

Now I can try your other tools did not work until now! :)

By the way, what is Comet mode?

Thx,
Gábor

bozimmerman, 2021-07-23:
Check out www.commodoreserver.com.  Comet Mode lets you access the services there.

Gabeki17, 2021-09-07:
Can Putty be used to connect?
I have no luck on configuring.

bozimmerman, 2021-09-07:
Since Putty doesn't support PETSCII, you won't have a good time at all.

bozimmerman, 2021-09-07:
If you need to connect from Windows, try CGTerm or SyncTerm.

Gabeki17, 2021-09-17:
> If you need to connect from Windows, try CGTerm or SyncTerm.

Aha!

Thank you very much! :)