bozimmerman/Zimodem #70: Feature request: Flush Buffer with DCD goes low.

dabonetn, 2021-06-30:
Issue: When the modem is in the middle of a file transfer and loses carrier, the computer get confused by no carrier but data is still incoming on the port.

Can we get the option to enable dumping the buffer, sending "NO CARRIER" and then returning to ready mode?

Thanks.

bozimmerman, 2021-07-11:
Yeesh, I thought I had already accounted for exactly this issue, but will look into it.
I need to know, though:
1. any flow control settings
2. Using the internal file transfer (Gurumodem) or transferring from a remote host?

dabonetn, 2021-07-11:
Flow control is usually CTS/RTS, 
Modem style is esp8266 with remote host.

bozimmerman, 2021-08-15:
I'm afraid I was unable to reproduce this.
Here's the steps I tried:
set to 1200 baud, RTS/CTS
ata6502
telneted to modem @ 6502, went into stream mode
from telnet client, pasted 865 chars, and then immediately closed telnet once characters started coming through on the modem serial side.
the modem successfully finished all 865 characters before NO CARRIER was sent and DCD was toggled back.

closed 2021-09-03 by bozimmerman