bozimmerman/Zimodem #71: Zimodem ouptuts weird numbers instead of "[LINE BUSY]" message.

ufud-org, 2021-07-12:
If I understand correctly, Zimodem will output a "[LINE BUSY]" message and hang up if there is already an incoming connection active and another client tries to connect via telnet. However, there seems to be a bug causing the output of numerous "3" characters instead of the message. Then the firmware hangs up, as is expected. (see attached screenshot).  

[image]

bozimmerman, 2021-07-15:
Hmm.. 3 is the NO CARRIER code, but I don't think that's what's happening.  I recently added the ability to customize the busy message, and now I'm wondering if some subsequent change broke that...
Thanks, will definitely look into this!

bozimmerman, 2021-08-15:
Fixed in 3.6.4 -- thanks for this report.

closed 2021-08-15 by bozimmerman

f4grx, 2021-08-23:
Thanks for the fix, the reported screnshoot is actually mine :)