bozimmerman/Zimodem #72: Crashes with current ESP8266 code

jcw, 2021-07-16:
I'm not able to build a proper version of the current code with the current Arduino & ESP8266:

```
C64Net WiFi Firmware v3.6.3
sdk=2.2.2-dev(38a443e) chipid=1335429 cpu@80
totsize=1024k ssize=341k fsize=139k speed=40m
INITIALIZED
READY.

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bf70 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000034 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 0190
3fffff70:  4020c360 3ffef968 3ffef964 3fff0c48
3fffff80:  3fffdad0 00000000 3ffef5ec 40206304
3fffff90:  3fffdad0 00000000 3ffef900 4020506a
3fffffa0:  feefeffe 00000000 3fff0c34 4020ab20
3fffffb0:  feefeffe feefeffe 3ffe8614 40100ca1
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

Some more build info:

```
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 311888          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27341   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1556  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 4344  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 31992 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 345129 bytes (40%) of program storage space. Maximum is 860144 bytes.
Global variables use 37892 bytes (46%) of dynamic memory, leaving 44028 bytes for local variables. Maximum is 81920 bytes.
```

It crashes as soon as a character is sent to the ESP8266 (1 Mb ESP-01)
Arduino 1.8.15, ESP8266 Boards 3.0.1, MacOS 11.4.

I did not succeed in installing the older 2.7.4 version of the ESP8266 package (probably a compiler toolchain mixup on my end), so it's not clear how to proceed from here. Is there an older binary image somewhere, so I could try that?

PS. This may be the same problem as #64, but I'm reporting it separately as this case is not NodeMCU-related.

dabonetn, 2021-07-16:
Here's a copy compiled with Arduino 1.8.13 with the 2.7.4 libs.

https://www.dabone.xyz/zmodem3.6.3.zip

ESP-01 selected, FS:64k OTA:~470K

jcw, 2021-07-16:
Many thanks! That one works fine:

```
C64Net WiFi Firmware v3.6.3
sdk=2.2.2-dev(38a443e) chipid=1335429 cpu@80
totsize=1024k ssize=372k fsize=51k speed=40m
INITIALIZED
READY.
ATH
OK
```

(Note: I uploaded using: `esptool.py write_flash --erase-all 0 zmodem3.6.3.bin`)

So yeah, it looks like the ESP 3.0.1 change is what broke the build.
I'll leave this issue open, as it's a good workaround for now.

arkwise, 2021-08-03:
> Here's a copy compiled with Arduino 1.8.13 with the 2.7.4 libs.
> 
> https://www.dabone.xyz/zmodem3.6.3.zip
> 
> ESP-01 selected, FS:64k OTA:~470K

do you have the firmware files for this build? (source)

dabonetn, 2021-08-07:
> do you have the firmware files for this build? (source)

I just pulled the source that day and compiled.

jcw, 2021-08-07:
To clarify: the only issue seems to be that the newer ESP 3.0.1 compiler toolchain doesn't work. Building Zimodem w/ ESP 2.7.4 solves this. I don't think the underlying cause for this issue has been determined.

arkwise, 2021-08-07:
@jcw OK cool I will try rolling back to the older versions and see if this helps :)

bozimmerman, 2021-08-15:
Thanks for figuring out that it is not safe, atm, to use head version of the esp8266 libs.  
At one time I specified the esp lib version to use, and then, one day, the head versions started working, so I changed the README to just say 'use current'.  And now, of course, I have to change the README back again to specify a version. <sigh>
Again, much thanks though.

closed 2021-08-15 by bozimmerman

Olifiers, 2021-12-22:
Folks, just a suggestion: I know that the Building section of the readme file mentions version 2.7.4, but wouldn't it be a good idea to add a note that newer versions do not work? I suggest this as I've lost a few hours scratching my head until I found this thread at last. Most people (like me) would assume the note referring to 2.7.4 was made at the time when such version was current, not imagining that newer versions won't work. Anyway... Just my two cents!