bozimmerman/Zimodem #74: Firmware crashes with exception on ESP8266

ufud-org, 2021-08-18:
Just flashed the new release on my ESP8266 but unfortunately, it causes an exacption (and crashes). See output from serial line below. My settings in Arduino IDE:

* generic ESP8266 module
* nonos-sdk-2.2.1-119
* flash all contents
* MMU: 32k cache + 32K iRAM

C64Net WiFi Firmware v3.6.4                                                                  
sdk=2.2.2-dev(a58da79) chipid=1458270 cpu@80                                                 
totsize=4096k ssize=343k fsize=51k speed=40m                                                 
INITIALIZED                                                                                  
READY.                                                                                       
                                                                                       
 --------------- CUT HERE FOR EXCEPTION DECODER ---------------                              
                                                                                             
Exception (0):                                                                               
epc1=0x40202f86 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000          
                                                                                             
>>>stack>>>                                                                                  
                                                                                             
ctx: cont                                                                                    
sp: 3ffffde0 end: 3fffffc0 offset: 0190                                                      
3fffff70:  4020c378 3ffef9b8 3ffef9b4 3fff0c98                                               
3fffff80:  3fffdad0 00000000 3ffef63c 40206320                                               
3fffff90:  3fffdad0 00000000 3ffef950 4020507e                                               
3fffffa0:  feefeffe 00000000 3fff0c84 4020ab38                                               
3fffffb0:  feefeffe feefeffe 3ffe8614 40100c9d                                               
<<<stack<<<                                                                                  
                                                                                             
--------------- CUT HERE FOR EXCEPTION DECODER ---------------                               
                                                              ����%%�                        
C64Net WiFi Firmware v3.6.4                                                                  
sdk=2.2.2-dev(a58da79) chipid=1458270 cpu@80                                                 
totsize=4096k ssize=343k fsize=51k speed=40m                                                 
INITIALIZED                                                                                  
READY.

ufud-org, 2021-08-19:
Problem solved, was related to the libraries I had installed. Nothing to do with the new release.

closed 2021-08-19 by ufud-org

allenhuffman, 2021-09-04:
What libraries did you have to switch to?

wx4cb, 2022-07-24:
I get the same here ... how did you fix it??? latest arduino ide, latest library installed via library and board managers