bozimmerman/Zimodem #75: Lost/corrupt packets during loading BBS pages using Telnet on real Commodore 64

Gabeki17, 2021-09-18:
Hi,

I have some issues on connecting to some BBS like "Theoldnet"
url: theoldnet.com
port: 6400

The issue description is that, some data is lost/corrupt during loading. ( I think simply not arriving)
For example it is always happenning that the "Theoldnet" banner/logo is always falling apart on the first load. (Some other bbs sites are loading with missing lines or words. (preatty random)

Also sometimes downloads are failing at the end. 
Download in reality is done on c64 side, but not telling the BBS server. (Protocol is Xmodem)

Terminal program: 
Using CCGMS 2021 on a real Commodore 64

configuration:

baud: 9600 (Does not seems to change any if lower like 1200)
full duplex
modem type: UP9600 / EZ232
Firmware: Zimodem
protocol: Xmodem

I am having the following wiring between my 60 and my ESP8266 D1 mini:

(DSR,DTR,RI is not connected "-|-" Is it the issue might be?)

# define DEFAULT_PIN_DSR 13 // D7 -|- L?
# define DEFAULT_PIN_DTR 12 // D6 -|- E?
# define DEFAULT_PIN_RI 14  // D5 -|- F?
# define DEFAULT_PIN_RTS 4  // D2 --- K
# define DEFAULT_PIN_CTS 5  // D1 --- D
# define DEFAULT_PIN_DCD 2  // D4 --- H
TX --- B - C - 7
RX --- M - 5

User port "jumper connections/connecetd together": 
B - C - 7
M - 5
L - 6

Thank you very much in advance!

All the best,
Gábor

Gabeki17, 2021-09-18:
I went forward a little with investigation.

I have done a littly spying on the rx/tx lines.

Simply connected SyncTERM also together with the 64. (Laptop/SyncTERM on the usb connector of D1 mini on one com5 port) While the 64 is connected as before.

I was able to have the same input for Syncterm. So is displaying the same.

Now. On syncterm I have no issue.

Might be. I don't really understand what is happening, but I think the issue is between the 64 and the esp8266.

So ESP8266 is receiving all right. But when forwarding it to the 64 something goes wrong.

With regards,
Gábor

Gabeki17, 2021-09-18:
Here is a video about: [https://www.youtube.com/watch?v=RPRbq7SDGHY](url)

Gabeki17, 2021-09-23:
I think it is a C64 side CCGMS 2021  issue.

It hangs at the begining.

If the modem is delayed for 7 sec no issue.

Also if I send 75 * 3 spaces at the begining of the connection, also good and this is actually a much faster workaround.

I know it is dirty to solve it on the modem side.

Wit hregards,
Gábor

bozimmerman, 2021-11-16:
Hello Gabor.  It sounds like this might be a flow control issue maybe?

Gabeki17, 2021-11-30:
> Hello Gabor. It sounds like this might be a flow control issue maybe? Watching your video, it would be my first suspect. Also, can you try another term, such as Novaterm?

Thank you!

I did try Nova. Also SyncTerm.
None of them are perfect. :) 
Also their phonebooks are terrible....

But anyways, basically I am having the most issues with all of them, when on 9600 baud.
Zmodem protocol works not. Y also not, nor x works reliable.
But when moving back to 2400 baud (different driver) It just works.
I have tested this with SyncTerm running on my Raspberry Pi.

Looks like the 9600 baud "trick" is not perfect.

With regards,
Gábor

bozimmerman, 2022-02-11:
The C64 and VIC-20 user ports are identical, so yea, that will work fine.
In fact, I ported all my little internet tools from the C-64 to the VIC.  Using IRC or FTP  in 22 cols may not sound fun, but it is possible!

Gabeki17, 2022-02-16:
> The C64 and VIC-20 user ports are identical, so yea, that will work fine. In fact, I ported all my little internet tools from the C-64 to the VIC. Using IRC or FTP in 22 cols may not sound fun, but it is possible!

Hi, I think a little different.

[image]

bozimmerman, 2022-02-16:
Wow.. I never knew that.  I never use pins 5-7 for serial, so it never occurred to me to look.  You should definitely not hook anything up to those pins.

Thanks for the correction.

Gabeki17, 2022-02-17:
> Wow.. I never knew that. I never use pins 5-7 for serial, so it never occurred to me to look. You should definitely not hook anything up to those pins.
> 
> Thanks for the correction.

Hello,

Are you using the modem at 2400 Baud?

The extra pins I think for the 9600 baud hack.

Regards,
Gábor

bozimmerman, 2022-02-23:
9600 baud hack on the VIC-20?   2400 baud?  I don't think we are talking about corrupt packets any more.

closed 2022-02-23 by bozimmerman

Gabeki17, 2022-02-24:
> 9600 baud hack on the VIC-20? 2400 baud? I don't think we are talking about corrupt packets any more.

No,no C64.

My interest is to use the same modem for both, the C64 and the VIC-20.
That is why i was to keep the wiring.