bozimmerman/Zimodem #78: Wifi does not re-connect after dropout

vk5zea, 2022-01-16:
I've been using Zimodem for a week or so now and I am noticing that it will not automatically re-connect to Wifi if the Access Point disappears temporarily. eg. rebooting a Wifi Router.

A similar thing happens if the WiFi Access Point is not present as Zimodem starts up. It will remain disconnected if the stored WiFi Access Point is enabled/starts after Zimodem boots.

Zimodem does not seem to check for the stored WiFi Access Point unless Zimodem is re-started.

Gabeki17, 2022-01-19:
> I've been using Zimodem for a week or so now and I am noticing that it will not automatically re-connect to Wifi if the Access Point disappears temporarily. eg. rebooting a Wifi Router.
>
> A similar [...]

Different versions of arduino addins are behaving little different.

Like in a different project for 2.7.4 as I remember had to write my own connection checking function.

Are you using that version might be?

With regards,
Gábor

bozimmerman, 2022-02-23:
I'm about 10% worried that retries might be seen as a bug if the modem becomes unresponsive to typing during the retry.  That said, it's a good idea to add this.

bozimmerman, 2022-05-17:
Finally adding this.

closed 2022-05-17 by bozimmerman