bozimmerman/Zimodem #80: Exception 0 on incoming byte

czarnikjak, 2022-02-17:
I have managed to build and flash Zmodem to ESP8266 Wemos D1 mini,

It boots up, but as soon as I send any character to it over uart, it crashes and reboots:

Zimodem Firmware v3.6.5
sdk=2.2.2-dev(38a443e) chipid=1458208 cpu@80
totsize=4096k ssize=341k fsize=1907k speed=40m
INITIALIZED
READY.

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40202f7e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 0190
3fffff70:  4020c37c 3ffef988 3ffef984 3fff0c68  
3fffff80:  3fffdad0 00000000 3ffef60c 40206318  
3fffff90:  3fffdad0 00000000 3ffef920 4020507a  
3fffffa0:  feefeffe 00000000 3fff0c54 4020ab3c  
3fffffb0:  feefeffe feefeffe 3ffe8614 40100c9d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------j

Gabeki17, 2022-03-13:
Switch to version 2.7.4

[image]

bozimmerman, 2022-04-27:
Gabeki17 nailed it.

closed 2022-04-27 by bozimmerman