bozimmerman/Zimodem #82: Zimodem Stream issue

barrow29, 2022-04-08:
I'm using a Feather HUZZAH esp 8266 board. I got it working for the most part but ran into this issue while trying to test the streaming before building the rest of the board. I was controlling the board via USB and TeraTerm I would open a new Teraterm window and connect to It again over Wi-Fi this worked for the most part but when using the window connected over Wi-Fi the USB window would not display the text until I had hit enter. In the other direction It worked exactly as I expected it to printing each character on the opposite window as I typed. If anyone knows what I might be missing or why this is happening your advice would be greatly appreciated.

barrow29, 2022-04-11:
After looking into it I believe this issue is being caused by the terminal programs I have attempted defaulting to "line at a time mode" Is the Zimodem supposed to send a WILL ECHO command when telnet is enabled or would that need to be added in?

bozimmerman, 2022-04-27:
DO/DON'T are part of the telnet protocol.  By *not* including the flag that turns on Telnet, you will avoid the behavior.

bozimmerman, 2022-05-17:
Adding a new flag ATS62 to disable the Telnet flag in the Dial command.

closed 2022-05-17 by bozimmerman