bozimmerman/Zimodem #86: Ring Indicator enhancement request

KC9LKE, 2022-09-21:
Hi Bo
First I’m using Zimodem and love it, thanks for your efforts!  
Back in the vintage days some used the RS232 Ring Indicator to wake up, turn on, their system so they could remote in and so that the system wasn’t running 24-7. When the remote session ended the RI would drop and the system would power down.  When I remote into Zimodem the RI stays on and then drops when the session disconnects, perfect, but now there is an issue.  When remote terminal apps or PC’s “sleep” they drop the connection and RI turns off.  This is not much of an issue for me but if I forget or I am allowing others to log in and they don’t modify their system settings my system could be rebooting or shutting down every couple of minutes, not good.
Would it be possible to add a timer setting in ATxx or Config for the RI output that’s settable from 0 to 255 minutes, 0=normal settings, so that if in Auto Answer mode and CD drops, RI will remain on for X minutes? If a connection is reestablished before timing out then the counter resets and only starts the next time CD drops.  This would greatly minimize the chance of the system rebooting every time the connecting app or PC “sleeps”.  Just a thought.  Thanks again.
Ted

bozimmerman, 2022-12-30:
So, the RI stands for "Ring Indicator" right, and so it indicates that the "Ring" signal is presently coming across your phone line.  That sounds like the signal should go active only during the brief limited audible phone "rings", and then stop.  That would mean it does not go active and then just stay active indefinitely.  May I ask why you aren't using DCD?  That sounds more like what you want...

Biker1bob, 2023-01-12:
yes I would agree, DCD means a remote connection to the "modem"  RI just means that someone is TRYING to call you.  The initial handshake as it were.

Biker1bob, 2023-01-15:
Actually now trying to use it, I see what is ment I think?  I have LEDs for both RI and DCD - when a call comes in things happen fast, so I am not sure how the programming is supposed to work, but RI does not turn on for a simple ping request.. which would issue a RING command it also does not turn on when trying to do a sustained telnet connection - lets say I set S0=10 .. SO each time a connection happens it will post a RING response and wait for the HOST to do something to answer - seems to be about 5+ secs. then issue another RING - I am guessing to simulate time between tones. But when this happens ON a Traditional MODEM the RI would be on as soon as it is getting something on the phoneline and then would turn off when DCD turns on.. I am not getting that. I do get DCD to turn on once the system answers after 10 rings.. but otherwise nothing.

KC9LKE, 2023-01-24:
Hi Bo
You are correct 
I will try DCD

closed 2023-02-08 by bozimmerman