bozimmerman/Zimodem #87: ATD / Path

czarnikjak, 2022-09-23:
Hi,
I am able to use ATD command to stream Gopher pages, for example like this:

ATD"gopherpedia.com:70"

However, I cannot access any resources or paths following / 
ATD"gopherpedia.com:70/about"
The above just streams the main page again, ignoring /about.

Is this a bug, or a feature request?

Thank you! 
BTW, Fantastic firmware!

bozimmerman, 2022-11-23:
Hmm.. ATD is just making a TCP connection, so the parser is only looking for Host and Port and that's all.  The /stuff is entirely ignored.  

Unfortunately, I don't know the gopher protocol.  The resource path probably must be sent as a command after the connection happens, like in HTTP.   I didn't even know gopher servers were still around -- adding a gopher client might be fun, sorta like we have an HTTP client with AT&G. :)

And -- damn that &G was already taken. :(

Thanks for the suggestion!

bozimmerman, 2024-02-28:
So, I ended up doing two things:
I added a gopher client for c64, c128, and vic-20.
I support gopher:// urls in at&g, wget, and other internet functions where http:// is supported

closed 2024-02-28 by bozimmerman