bozimmerman/Zimodem #88: Can this be used with an actual modem on the serial side ?

M0NWY, 2022-10-17:
Hello, I'm the owner of a dreamcast that can only connect to the internet with an actual modem. 
I've recently made it work with an old pc and a modem but I'd prefer the more elegant solution of using a esp32 as a wifi to modem bridge.
It looks like you've done 99.7% of the work necessary for this to happen, so will this firmware answer rings from the serial side ?

bozimmerman, 2022-11-23:
Yes, it will.  The firmware supports BBS type connection messages.  In fact, I run a BBS on a Commodore 128 using it.  Does it really need that, though?  I suspect the dreamcast would be trying to dial out, not receive incoming calls, right?  If so, that definitely works. :)

closed 2022-11-23 by bozimmerman

M0NWY, 2022-11-23:
Thanks for getting back to me, yes the dreamcast is calling out but you
can't intercept the serial lines as the modem is directly on the system bus
and a lot of games are written to interact with it directly. So the use of
an old modem is a reasonable solution. Thanks again, I'll be experimenting
over Xmas!