bozimmerman/Zimodem #89: Exception 9

mteletin, 2022-12-01:
I compiled the code with Arduino 2.0.2 and uploaded on an ESP-12e with success however, immediately after sending any command I get Exception 9 - unaligned address. Any idea on why do I get this? I mean, I understand the problem and what the exception means but... shouldn't this code just compile and run without this issue? Am I using the wrong (too new) Arduino IDE/compiler?
If I enable warnings on the compiler... it's full of them and I can't compile anything.
Any help is appreciated (even suggesting a different SDK to build the code and uploading).

mteletin, 2022-12-01:
Just compiled it under 1.6.13... same problem with Exception 9. I toyed with compiler options to no avail.

mteletin, 2022-12-01:
[image]

ytmytm, 2022-12-04:
I had a similar problem today. In my case it was solved by downgrading ESP8266 package (boards) from version 3.0.x to 2.7.4 as suggested in #74

mteletin, 2022-12-04:
Haven't tried this, but on a different project I had to downgrade and it worked. I assume it's the same with zimodem. Perhaps we can close this one.

closed 2022-12-04 by mteletin