bozimmerman/Zimodem #98: Amiga compatibility question

Gabeki17, 2023-02-17:
Hello,

I am a proud user on my C64!

But is it also compatible with my Amiga might be?

By my expectation it is, but not finding a wiring diagram for.

Thx,
Gábor

Biker1bob, 2023-02-17:
For sure it is.  Your Amiga will be using a std rs232 25pin port. You will want to use the std firmware version.. not the c64 version, and wire the signals up to std null modem connection to the port of the modem. Zimodem firmware will work with any rs232 computer.

Gabeki17, 2023-02-17:
> For sure it is. Your Amiga will be using a std rs232 25pin port. You will want to use the std firmware version.. not the c64 version, and wire the signals up to std null modem connection to the port of the modem. Zimodem firmware will work with any rs232 computer.

Thank you!
Is there an std version? Or you mean, i need to set a define in the code?

Biker1bob, 2023-02-17:
CBM8bit as I understand it is the C64 version, and or specific settings required for the proprietary serial of the 64 - the STD Zimodem 3.7.1 (current version) is for all RS232 serial computers.

bozimmerman, 2023-02-27:
This is the way I would put it:
Whether the firmware works on the Amiga or C64 depends on the hardware.  
If the hardware uses TTL voltage levels for all the serial lines, then it's for the C64, VIC, C128 (others?).
If it uses standard RS-232 voltage levels, then it's for the Amiga (or PC, or CBM-II, or lots and lots of machines that uses standard RS-232).
You can buy a standard RS-232 device called the "Gurumodem".  I googled it and found one on sale here:
https://www.retrocomputershopper.com/store/Gurumodem-p134166147
There might be others?

closed 2023-02-27 by bozimmerman

Gabeki17, 2023-03-12:
> This is the way I would put it: Whether the firmware works on the Amiga or C64 depends on the hardware. If the hardware uses TTL voltage levels for all the serial lines, and the right edge connector and pinout, then it's for the C64, VIC, C128 (others?). If it uses standard RS-232 voltage levels, and a DB9 or DB25 connector, then it's for the Amiga (or PC, or CBM-II, or lots and lots of machines that uses standard RS-232). You can buy a standard RS-232 device called the "Gurumodem". I googled it and found one on sale here: https://www.retrocomputershopper.com/store/Gurumodem-p134166147 There might be others?
> 
> (P.S. I use an Amiga 2000HD for most of my Zimodem testing, and I use it on an Amiga 1000 because it's the only way to network on it -- so heck yea it works on Amiga).

Thank you!

Do you have a wiring diagram might be?

With regards,
Gábor