bozimmerman/Zimodem #102: 3.7 firmware and bbs host, echo enabled by default?

dabonetn, 2023-03-08:
I had to figure out what had broken Centipede BBS support for my fork, and it ended up being a BBSHost echo being enabled by default.

Is this supposed to be the case? or should it be off be default.

If you have chosen it to be on by default, could you give me a hint on where in the source code to change it to default off?

Thanks.

bozimmerman, 2023-03-22:
Hi Dabo -- 
It's not clear to me where the echoing is happening.
Do you mean in command mode, where you enter AT commands?  Or do you mean after the BBS host connection, in Stream mode?

The former can be disabled with "ATE0".
The later is disabled by default, *unless* you include an "E" in the "ATA" command.   For example "ATAE4444".

dabonetn, 2023-03-22:
It's echoing back to the host while in stream mode.

Under enable bbs mode menu in at+config , you have an option to enable echo. When echo is enabled, everything the bbs sends is echoed directly back to the bbs software.

Easy to test, enable bbs mode, and connect from another machine. When you type in the terminal on the host while connected, it echos the local characters to the screen. When the echo is set to off in the bbs mode setup in at+config, no echo occurs.

closed 2023-03-26 by bozimmerman