bozimmerman/Zimodem #103: Can't compile: error: no return statement in function returning non-void [-Werror=return-type]

acejacek, 2023-03-25:
Arduino IDE  throws back few errors like that in wificlientnode.ino, zcommand.ino and probably few others

bozimmerman, 2023-03-26:
Wow -- no idea why I've never seen those errors, or why they haven't caused runtime crashes.  Weird, right?
Anyway, the fix in WiFiClientNode I could see, but the changes to zimodem.ino are useless because of .. I dunno, EOL changes?  It's just showing the entire file as changed. 

Any way, you can comment here with a local diff of that file, or some way that I can visually see the changes?

closed 2023-03-26 by bozimmerman

acejacek, 2023-03-27:
That was super simple change, done directly in github.com. Added line 1871 with ‘return ZERROR’.” At the end of doHangupCommmand  No idea why diff looks so strange. Didn’t verify it prior, sorry. Maybe some collisions?BR/Jacek