bozimmerman/Zimodem #108: Arduino sys/socket.h: No such file or directory

badre2911, 2023-04-25:
Hi,
i want to compil Zimodem on arduino 1.8.19,
board is nodemcu 8266 
Arduino board version is 3.1.2 but i have  this error

```
In file included from D:\donnees\dev\ARDUINO\sketch\ESP8266\arduino\Zimodem-master\zimodem\src\libssh2\bcrypt_pbkdf.c:21:
D:\donnees\dev\ARDUINO\sketch\ESP8266\arduino\Zimodem-master\zimodem\src\libssh2\libssh2_priv.h:108:11: fatal error: sys/socket.h: No such file or directory
  108 | # include <sys/socket.h>
      |           ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

on windows sys/socket.h not exist, is for linux.

Any idea ?
It's possible to compil to arduino on windows 10 ?
Thanks.

badre2911, 2023-04-25:
I found the problem, it's because 3.1.2 is not compatible with 2.7.4,
maybe one day you will upgrade the code.

badre2911, 2023-04-26:
Where can i have zimodem  pre-build binary 4.0 for esp82266 ?
Thanks in advance.

bozimmerman, 2023-05-25:
This should be fixed now.

closed 2023-05-25 by bozimmerman