bozimmerman/Zimodem #113: brand new compile, undefined reference

wx4cb, 2023-10-18:
just cloned the repo and trying to compile it for an 8266 module using the latest arduino IDE and i get the following error at the end of the compile and not sure how to proceed. I did do a search in the history of the issues but couldnt find anything.

when i "opened" the zimodem.ino file it opened a bunch of other inos so it should have everything it needs no ?

```
FQBN: esp8266:esp8266:generic
Using board 'generic' from platform in folder: /home/charlesb/.arduino15/packages/esp8266/hardware/esp8266/3.1.2
Using core 'esp8266' from platform in folder: /home/charlesb/.arduino15/packages/esp8266/hardware/esp8266/3.1.2

<CUT>

/home/charlesb/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/zimodem.ino.cpp.o: in function `_ZN6ZPrint18switchToPostScriptEPc':

/usr/src/Zimodem/zimodem/zprint.ino:119: undefined reference to `_ZN9ZSLIPMode8SLIP_ENDE'

/home/charlesb/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /usr/src/Zimodem/zimodem/zprint.ino:121: undefined reference to `_ZN9ZSLIPMode8SLIP_ESCE'

collect2: error: ld returned 1 exit status

Using library ESP8266WiFi at version 1.0 in folder: /home/charlesb/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi 
exit status 1

Compilation error: exit status 1

````

bozimmerman, 2023-10-19:
Thanks!  SLIP is definitely not done, so all I did was re-disable the include flag, to prevent it from trying to compile.

closed 2023-10-19 by bozimmerman

bozimmerman, 2023-10-19:
Update your repo and try again

wx4cb, 2023-10-21:
> Update your repo and try again

i just did that.... in fact i completely re-cloned it from scratch and still get the same error

```
/home/charlesb/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gcc -fno-exceptions -Wl,-Map -Wl,/tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/zimodem.ino.map -g @/home/charlesb/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/warnings/none-cflags -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static -L/home/charlesb/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/lib -L/home/charlesb/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/lib/NONOSDK22x_190703 -L/tmp/arduino/sketches/0D278369A9DE032FE150649741253A01 -L/home/charlesb/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/sdk/libc/xtensa-lx106-elf/lib -Tlocal.eagle.flash.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/zimodem.ino.elf -Wl,--start-group /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/zimodem.ino.cpp.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/bcrypt_pbkdf.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/blowfish.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/channel.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/comp.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/crypt.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/global.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/hostkey.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/keepalive.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/kex.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/mac.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/mbedtls.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/misc.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/packet.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/pem.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/session.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/transport.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/userauth.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/libssh2/userauth_kbd_packet.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/src/missing/slipif.c.o /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/libraries/ESP8266WiFi/ESP8266WiFi.a /tmp/arduino/cores/b2e70222f8ade4aa68987310d6679daa/core.a -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group -L/tmp/arduino/sketches/0D278369A9DE032FE150649741253A01
/home/charlesb/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/0D278369A9DE032FE150649741253A01/sketch/zimodem.ino.cpp.o: in function `_ZN6ZPrint18switchToPostScriptEPc':
/usr/src/Zimodem/zimodem/zprint.ino:119: undefined reference to `_ZN9ZSLIPMode8SLIP_ENDE'
/home/charlesb/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /usr/src/Zimodem/zimodem/zprint.ino:121: undefined reference to `_ZN9ZSLIPMode8SLIP_ESCE'
collect2: error: ld returned 1 exit status
````

bozimmerman, 2023-10-23:
Hi.  Since I can't recreate this, I can only guess:
Notice that the Sketch ID 0D278369A9DE032FE150649741253A01 is identical in both compiles.  If the Arduino compiler really saw your second attempt as a different Sketch, I would think that it would generate a new ID, and thus compile from a different location.  

So, try removing your /tmp/arduino/sketches/0D... directory.

The real question though -- can you look in zprint.ino and tell me if the you see any reference to SLIP?  There isn't!  It's really bizarre.