bozimmerman/Zimodem #117: build error, variables not declared in this scope (DEFAULT_DCD_HIGH)

anon905, 2023-12-12:
Trying to build the firmware according to the instructions in the readme, I got an error that some variables were "not declared in this scope" in zcommand.ino. Those variables seem to be the ones with name changes in a recent commit. I was able to build the firmware with the release previous to that commit, and I was also able to build the firmware by changing the variable names to the new ones (HIGH -> ACTIVE), but I've not yet put it onto a board to make sure it actually works. Did some of the variable names not get changed, or am I having some other issue?

Here's the build error I get with the latest release:
`/Zimodem/zimodem/zcommand.ino: In member function 'ZResult ZCommand::doInfoCommand(int, uint8_t*, int, bool)':
zcommand:737:22: error: 'DEFAULT_DCD_HIGH' was not declared in this scope
     if((dcdActive != DEFAULT_DCD_HIGH)||(dcdInactive == DEFAULT_DCD_ACTIVE)||(showAll))
                      ^
zcommand:741:22: error: 'DEFAULT_CTS_HIGH' was not declared in this scope
     if((ctsActive != DEFAULT_CTS_HIGH)||(ctsInactive == DEFAULT_CTS_ACTIVE)||(showAll))
                      ^
zcommand:745:22: error: 'DEFAULT_RTS_HIGH' was not declared in this scope
     if((rtsActive != DEFAULT_RTS_HIGH)||(rtsInactive == DEFAULT_RTS_ACTIVE)||(showAll))
                      ^
zcommand:749:21: error: 'DEFAULT_RI_HIGH' was not declared in this scope
     if((riActive != DEFAULT_RI_HIGH)||(riInactive == DEFAULT_RI_ACTIVE)||(showAll))
                     ^
zcommand:753:22: error: 'DEFAULT_DTR_HIGH' was not declared in this scope
     if((dtrActive != DEFAULT_DTR_HIGH)||(dtrInactive == DEFAULT_DTR_ACTIVE)||(showAll))
                      ^
zcommand:757:22: error: 'DEFAULT_DSR_HIGH' was not declared in this scope
     if((dsrActive != DEFAULT_DSR_HIGH)||(dsrInactive == DEFAULT_DSR_ACTIVE)||(showAll))
                      ^
exit status 1
'DEFAULT_DCD_HIGH' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

vk5ln, 2024-01-07:
I have noticed this as well with the recent update. I use a zimodem.ino file that is slightly modified to allow Zimodem to work with the LEDs on a Sven Petersen WiFi Modem, this breaks things even more. When I change the lines in zimodem.ino directly related to the LED pin changes I still can't get it to build without errors.

bozimmerman, 2024-02-28:
I'm really sorry this happened, and am sure you found it suddenly fixed later on.

closed 2024-02-28 by bozimmerman