bozimmerman/Zimodem #133: how to disable ATT debug?

varna9000, 2024-05-15:
Hi, first of all, thank you for Zimodem :)

I'm struggling to find how to disable the byte array which is returned by ATT"[MESSAGE]". It looks something like this: 
`[ 130, 17, 45 ]`  and is being echoed before any serial message. ATV0, ATX0 and ATQ0 doesn't seem to help. I'm building the firmware from the current master on esp-01.

bozimmerman, 2024-06-23:
Huh.
So, the [ 130 17 45] appears to be the packet headers you see in Command mode after connecting to a remote host.
The ATT"MESSAGE" command does not normally give any response other than 0K or ERROR or whatever, unless you intentionally turn on CRC8 mode, in which case it returns a single number.

IOW, I'm not quite sure what's going on.  Can you paste in a transaction log that starts from modem initialization until you issue the ATT command and see [130 17 56] type stuff?  Thanks.

varna9000, 2024-08-12:
I spent some more time on this today. It seems I've misunderstood or rather assumed wrongly how things work. I'm attaching a sample session.

1. I create new server listener on esp01
2. I connect with telnet from my laptop to it
3. Here I made two wrong assumptions:
   - The connection will not go into streaming mode and
   - the newly created connection is made "current one"
 
4. I send `ATT"test"` and after sending the connection goes automatically in streaming, where all text sent from telnet returns the said array with three hex values.
5. If I disable auto streaming with `ATS41=0`, then `ATT"test"` returns `ERROR`
6. So finally I select the connection as "current" with `ATCn`, and only then ATT"test" sends the string without giving back error or going into streaming mode.

[image]

closed 2024-09-04 by bozimmerman