bozimmerman/Zimodem #134: Latest firmware VIA the online update function

mamejay, 2024-06-10:
Hi Bo,
Will this latest firmware be available VIA the AT update command direct to the modem?
I just tried and 3.7.1 is the latest online version and not 4.0.1

Thanks

mamejay, 2024-06-10:
Further to this I tried to install the latest firmware manually, but I get no response from the modem.  No activity LED or anything from the ESP-12F device.  Rolling back sorts out everything.  Is this 4.0.1 firmware changed now and no longer working with legacy devices?

bozimmerman, 2024-06-23:
The old ESP-8266 based firmware is now on a different branch (the 371 branch).  The 4.x (master) branch is for the next rev modems that use 8MB ESP-32 chips (mostly).

So! Please try building the 371 branch on github.  That should work better for you.

closed 2024-06-23 by bozimmerman

mamejay, 2024-06-23:
Thank you so much!!  I have just performed the online update and left it as that.