bozimmerman/Zimodem #145: LED strip support

timotto, 2024-08-29:
I am building a Zimodem for my Apricot F1 computer which I plan to put into an actual modem case. I want to re-use the modem's front panel with an WS2812B LED strip to show status and activity.

Are you interested in a pull request to support individually addressable LED strips in Zimodem? It will however add a dependency to Zimodem, most likely FastLED.

I would add a a new class with a loop method to flash LEDs on RX/TX activity, while the other LEDs are status LEDs similar to the existing LED code. I would add configuration code, so a user can define at runtime the type, order, color, and functions of LEDs, as it differs between modem models. I'd re-use a GPIO pin from the existing LED code, as it is unlikely to run both at once. Due to the addressable LEDs, only one pin is required to control all the LEDs.

With this pull request I can avoid running a fork, and if the functionality will be enabled by default for ESP32 I would profit from OTA updates in the future.

frnno967, 2024-08-29:
Scott from Tattler Solutions has a fork of Zimodem that might be what you want. His website is here: https://tempestfpga.com/

closed 2024-09-04 by bozimmerman

wx4cb, 2024-09-21:
> Scott from Tattler Solutions has a fork of Zimodem that might be what you want. His website is here: https://tempestfpga.com/

pretty cute, wonder if the source for the software is available on github? :smile:

timotto, 2024-09-24:
If you look at the pictures of the PCB it looks like the LEDs for data and flow control are driven by the actual serial data lines. I believe the source code of the fork is at https://github.com/na103/zimodem as it supports a DAC to play the dial up and hand shake sounds.