bozimmerman/Zimodem #147: Compile error on New build for esp32

wx4cb, 2024-09-21:
just found this project and wanted to put it on a spare ESP32 board i have, however i'm running into this compilation error in the arduino IDE (latest version of code, libraries and IDE - fresh install of all on a new laptop). Any assistance appreciated. the board is just a regular generic esp32-wroom dev board.

```
/usr/src/Zimodem/zimodem/zimodem.ino: In function 'void setHostName(const char*)':
/usr/src/Zimodem/zimodem/zimodem.ino:320:34: error: 'TCPIP_ADAPTER_IF_STA' was not declared in this scope
  320 |       tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, hname);
      |                                  ^~~~~~~~~~~~~~~~~~~~
/usr/src/Zimodem/zimodem/zimodem.ino:320:7: error: 'tcpip_adapter_set_hostname' was not declared in this scope
  320 |       tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, hname);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1

Compilation error: 'TCPIP_ADAPTER_IF_STA' was not declared in this scope

```

SanguineBrah, 2024-09-22:
Good timing, as I was trying to build for a WROOM module this afternoon. It doesn't work because you are using the latest versions of the IDE and libraries. I managed to build by installing the specific versions mentioned in the readme - IDE version 1.8.10, installing the espressif ESP32 package in board managers version 2.08 (2.07 won't install for me) and choosing "ESP32 Dev Module" as the board. You will want to comment out some includes as needed e.g. INCLUDE_SD_SHELL.

wx4cb, 2024-09-22:
> Good timing, as I was trying to build for a WROOM module this afternoon. It doesn't work because you are using the latest versions of the IDE and libraries. I managed to build by installing the specific versions mentioned in the readme - <CUT>

Interesting.. I dont think ide version would be an issue due to it not being the compiler, in fact it just compiled with the latest IDE no problem. however having to use a esp library/compiler that's 15 versions behind is a little worrysome to some degree to me.

petrkr, 2024-10-27:
It is because it uses deprecated 3.x ESP-IDF SDK functions.

https://www.esp32.com/viewtopic.php?t=13205

If you will downgrade Arduino ESP32 SDK back to 2.x (which use ESP IDF 4.x), then it still works (Just tested latest 2.0.17).

closed 2025-04-21 by bozimmerman